adafruit/Adafruit_nRF52_Arduino #538: Allow user to easily increase size of Serial Buffer during setup of Serial

thaanstad, 2020-07-15:
I have a function called processSerialCommand() that reads data from the serial port and then uses serial comparisons to process the desired functions.  It utilizes some data parsing with semicolons to get the data as required.  It isn't very efficient and it is also very memory intensive but on the other hand it is very easy to use and read.  When passing a serial command from the computer and the command is large enough the serial buffer on the NRF52832 isn't large enough to capture the whole command and the data is lost.  I was able to increase the SERIAL_BUFFER_SIZE in the core (RingBuffer.h) from 64 to 256 and since then haven't had issues.  I am interested in either increasing the size of this in the core or allowing the user to change it to their needs during setup possibly prior to or during Serial.begin(). 

//simple way to pass serial commands to MCU.  command format is: command;data;data;  (up to 8 data entries)
//The user can retrieve the values as floats or as strings as shown in the examples
bool processSerialCommand(Stream *ptrSer)
{
	if (ptrSer->available())
	{
		delay(100);
		char prevChar;
		char inputCharacter;
		String inputData;

		while (ptrSer->available())
		{
			inputCharacter = ptrSer->read();
			inputData.concat(inputCharacter);

			if (inputCharacter == '\0' && prevChar == ';')// look for the null character instead of double semicolon
			{
				break;
			}
			else
			{
				prevChar = inputCharacter;
				delay(1);
			}
		}

		//Serial.println("");
		//Serial.println(inputData);

		//inputData = inputData.substring(0, inputData.length() - 1);
		int commandLength = 0;

		for (int i = 0; i < inputData.length(); i++)
		{
			if (inputData.charAt(i) == ';')
			{
				commandLength = i;
				break;
			}
		}
		String command = inputData.substring(0, commandLength);
		String data = inputData.substring(commandLength + 1, inputData.length());
		String values = data;

		//Serial.println(values);
		//Serial.println(commandLength);
		//Serial.println(inputData.length());
		float value[20] = { 0 };
		String stringValues[20] = { "" };

		int valueLength = 0;
		for (int j = 0; j < 20; j++)
		{
			for (int i = 0; i < values.length(); i++)
			{
				if (values.charAt(i) == ';')
				{
					valueLength = i;
					break;
				}
			}
			value[j] = (values.substring(0, valueLength)).toFloat();
			stringValues[j] = (values.substring(0, valueLength));
			//Serial.print(value[j]);
			//Serial.print("; ");
			values = values.substring(valueLength + 1, values.length());
			//Serial.println(values);
			valueLength = 0;
		}

		ptrSer->println();

		ptrSer->print(command);
		ptrSer->print(", ");

		if (command.equals("stringTest"))//stingTest;hello;test;123;
		{
			for (int i = 0; i < 20; i++)
			{
				ptrSer->print(stringValues[i]);
				ptrSer->print(", ");
			}
			ptrSer->println("");

			return true;
		}
		if (command.equals("floatTest"))//floatTest;30.4;55.6;70008.0;
		{
			for (int i = 0; i < 20; i++)
			{
				ptrSer->print(value[i]);
				ptrSer->print(", ");
			}
			ptrSer->println("");

			return true;
		}else{}
       return false;
}

Perhaps there is another solution but I would be interested in being able to define the size of the serial buffer per my projects requirements.  Something like Serial.setBuffer(256).  Perhaps the ringbuffer class could have a new constructor that allows allocating this size.  In the Uart class the rxBuffer could be pointer to a RingBuffer object and then it would be easy to define a new RingBuffer object with this increased buffer size.  If this seems reasonable to others I can try to implement and test it otherwise I am open to other suggestions as to how this would best be implemented.

hathach, 2020-07-15:
yeah, increasing serial buffer can be useful for some project. Do you mind submitting a PR for this ?

hathach, 2020-07-16:
@thaanstad since you label this as feature, please update your post to use the feature template

huddy987, 2020-07-25:
I’d be interested in implementing this feature. I think the thing that makes the most sense is to add a second optional argument to serial.begin since this buffer probably shouldn’t be changed after serial has been opened

hathach, 2020-07-27:
@huddy987 It should have its own API `setBufferSize(tx, rx)` . It can be a bit complicated since USB CDC fifo is part of tinyusb config file. The hardware UART on 832 can probably easier. 

PS: this issue is closed since OP failed to response to my request. You could open an separated issue using the feature template.

closed 2020-07-27 by hathach