adafruit/Adafruit_nRF52_Arduino #539: Changing MTU not possible

pythonimus, 2020-07-15:
**Describe the bug**
When reading from a Characteristic only the first 20 bytes are read. When attampting to change the MTU to read larger Characteristics the MTU is not actually changed.

When I invoke the change MTU function just like in the example noting happens:
Serial.println(conn->getMtu());
conn->requestPHY();
conn->requestDataLengthUpdate();
conn->requestMtuExchange(300);
 conn->requestConnectionParameter(6);
delay(1000);
Serial.println(conn->getMtu());

output of the console ist just:
23
23

The MTU is never changed.

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.13 on Windows 10
 - **BSP**   : 0.20.5
 - **Board** : Feather nRF52840 Express
 - **Sketch**: Bluefruit52Lib\examples\Central\central_custom_hrm merged with Bluefruit52Lib\examples\Central\central_throughput

**To Reproduce**
Steps to reproduce the behavior:
1. Get an app on Android which has Characteristics over 20 bytes and expects you to change the MTU.
2. Change MTU and read from Characteristic

hathach, 2020-08-03:
Can you try to have another feather running as peripheral, the Android app may not support and/or handle the MTU negotiation request. Also please provide the debug log with level = 2 as well.

hathach, 2020-09-09:
closed due to lack of user response.

closed 2020-09-09 by hathach

ProgrammingCube, 2020-09-15:
I have found a solution!
The default max length of MTU is 23. To change this, change line 64 in the file:

Arduino15/packages/adafruit/hardware/nrf52/0.21.0/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include/ble_gatt.h

or equivalent. You should see a define of 23.