adafruit/Adafruit_nRF52_Arduino #544: Bluefruit.begin() and USB Device Not recognized.

innoagg, 2020-07-28:
Compiling with the current master branch the USB device is not recognized. An empty sketch calling Bluefruit.begin() in the setup is all that is required to reproduce. 

```
#include <stdio.h>
#include <Arduino.h>
#include <bluefruit.h>

void setup() {
  Serial.begin(115200);
  Bluefruit.begin(0, 4);
}

void loop() {
  
}

```

Boards: pca10059, pca10056
Windows 10

[image]

hathach, 2020-07-29:
I confirmed the issue, there is still issue with portYield() in ISR https://github.com/hathach/tinyusb/pull/468 and the hardware initialization. Initialize it late cause missing initial setup packet. I am troubleshooting it now.

hathach, 2020-07-31:
This really give me a headache now, I need to do commit bisecting to figure the reason.

innoagg, 2020-07-31:
Wow, thank you for all your efforts. Much appreciated. Try to enjoy your weekend!

hathach, 2020-08-01:
should be fixed by #546 but need https://github.com/hathach/tinyusb/pull/478 to be merged first

PR #546 (2020-08-04): fix power usbready not handled if bluefruit begin() is called fast enough
- fix POWER USB Ready may not  be handled when Bluefruit.begin() is invoked fast enough.
- fix #544 
- need https://github.com/hathach/tinyusb/pull/478 to be merged first then bump up the tinyusb core.
- Also clean up stuffs as well.
closed 2020-08-04 by hathach