adafruit/Adafruit_nRF52_Arduino #564: MAC address change

willwade, 2020-08-27:
**Is your feature request related to a problem? Please describe.**
We are making a device for people who are disabled who use morse code to write - and in built scanning to control a cursor. On iOS this is switch scanning. It listens to a ble keyboard. The problem is if iOS has one ble keyboard setup for scanning typing is ignored ; only the keys for scanning are listened to. So to use both typing in morse and scanning you have to have two ble devices. If you change device name/manufacturer name etc then it does nothing. iOS sees each device as unique only by the MAC address. So instead of having two ble devices it would be far easier if we had two MAC address (or a way to change it) 

**Describe the solution you'd like**
A Way of changing the MAC address - or swapping to another predefined MAC address (does this help ? https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s132.api.v3.0.0%2Fgroup___b_l_e___g_a_p___f_u_n_c_t_i_o_n_s.html&anchor=gaef69dc212534adf4b78d211edce2267b)

**Describe alternatives you've considered**
Altering manufacturer, device name. Does nothing for iOS to see it as a different device. 

**Additional context**
Add any other context or screenshots about the feature request here.

jpconstantineau, 2020-08-27:
Is this functionality already taken care of by  AdafruitBluefruit::setAddr (
ble_gap_addr_t* gap_addr) ?

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/src/bluefruit.cpp#L527

hathach, 2020-08-28:
> Is this functionality already taken care of by AdafruitBluefruit::setAddr ( ble_gap_addr_t* gap_addr) ? https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/src/bluefruit.cpp#L527
> […](#)
> -- Pierre

Yeah, that is the function, thank you for answering the question.

hathach, 2020-09-09:
closed due to lack of activities.

closed 2020-09-09 by hathach