adafruit/Adafruit_nRF52_Arduino #565: Data is lost when data is sent to the PC by Serial.println.

arms22, 2020-08-28:
**Describe the bug**
Data is lost when data is sent to the PC by Serial.println.

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.13 on Windows 10
 - **BSP**   : 0.20.5
 - **Board** : Feather nRF52840 Express
 - **Sketch**: 
```
void setup() {

  // put your setup code here, to run once:
  Serial.begin(115200);
  while ( !Serial ) delay(10);   // for nrf52840 with native usb

  Serial.println("Bluefruit52 USB Serial Test");
  Serial.println("-----------------------------------\n");

  uint32_t t2 = millis() + 5000;
  uint32_t wrsize = 0;
  SemaphoreHandle_t mutex = xSemaphoreCreateMutex();
  while (millis() < t2) {
    if (xSemaphoreTake(mutex, portMAX_DELAY) == pdTRUE)
    {
      wrsize += Serial.println("AaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaCcccBbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb");
      xSemaphoreGive(mutex);
    }
  }
  vSemaphoreDelete(mutex);
  delay(1000);
  Serial.print(wrsize / 5);
  Serial.println("bytes/Sec");
}

void loop() {
}
```

I think that overwriting occurs when writing data to the endpoint buffer.
Please check the patch.

```
diff --git a/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/cdc/cdc_device.c b/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/cdc/cdc_device.c
index dcdf56a..f551c14 100644
--- a/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/cdc/cdc_device.c
+++ b/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/cdc/cdc_device.c
@@ -58,6 +58,8 @@ typedef struct
 #if CFG_FIFO_MUTEX
   osal_mutex_def_t rx_ff_mutex;
   osal_mutex_def_t tx_ff_mutex;
+  osal_mutex_def_t mutex_def;
+  osal_mutex_t mutex;
 #endif
 
   // Endpoint Transfer buffer
@@ -161,16 +163,19 @@ uint32_t tud_cdc_n_write_flush (uint8_t itf)
 {
   cdcd_interface_t* p_cdc = &_cdcd_itf[itf];
 
+  osal_mutex_lock(p_cdc->mutex, OSAL_TIMEOUT_WAIT_FOREVER);
+
   // skip if previous transfer not complete yet
-  TU_VERIFY( !usbd_edpt_busy(TUD_OPT_RHPORT, p_cdc->ep_in), 0 );
+  TU_VERIFY_HDLR( !usbd_edpt_busy(TUD_OPT_RHPORT, p_cdc->ep_in), 0, osal_mutex_unlock(p_cdc->mutex) );
 
   uint16_t count = tu_fifo_read_n(&_cdcd_itf[itf].tx_ff, p_cdc->epin_buf, TU_ARRAY_SIZE(p_cdc->epin_buf));
   if ( count )
   {
-    TU_VERIFY( tud_cdc_n_connected(itf), 0 ); // fifo is empty if not connected
+    TU_VERIFY_HDLR( tud_cdc_n_connected(itf), 0, osal_mutex_unlock(p_cdc->mutex) ); // fifo is empty if not connected
     TU_ASSERT( usbd_edpt_xfer(TUD_OPT_RHPORT, p_cdc->ep_in, p_cdc->epin_buf, count), 0 );
   }
 
+  osal_mutex_unlock(p_cdc->mutex);
   return count;
 }
 
@@ -206,6 +211,7 @@ void cdcd_init(void)
 #if CFG_FIFO_MUTEX
     tu_fifo_config_mutex(&p_cdc->rx_ff, osal_mutex_create(&p_cdc->rx_ff_mutex));
     tu_fifo_config_mutex(&p_cdc->tx_ff, osal_mutex_create(&p_cdc->tx_ff_mutex));
+    p_cdc->mutex = osal_mutex_create(&p_cdc->mutex_def);
 #endif
   }
 }
```

arms22, 2020-08-28:
There was a bug in the above patch and it has been fixed.

```
diff --git a/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/cdc/cdc_device.c b/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/cdc/cdc_device.c
index dcdf56a..92ac6d5 100644
--- a/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/cdc/cdc_device.c
+++ b/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/cdc/cdc_device.c
@@ -58,6 +58,9 @@ typedef struct
 #if CFG_FIFO_MUTEX
   osal_mutex_def_t rx_ff_mutex;
   osal_mutex_def_t tx_ff_mutex;
+  osal_mutex_def_t mutex_def;
+  osal_mutex_t mutex;
+  uint8_t busy;
 #endif
 
   // Endpoint Transfer buffer
@@ -162,6 +165,7 @@ uint32_t tud_cdc_n_write_flush (uint8_t itf)
   cdcd_interface_t* p_cdc = &_cdcd_itf[itf];
 
   // skip if previous transfer not complete yet
+  TU_VERIFY( !p_cdc->busy, 0 );
   TU_VERIFY( !usbd_edpt_busy(TUD_OPT_RHPORT, p_cdc->ep_in), 0 );
 
   uint16_t count = tu_fifo_read_n(&_cdcd_itf[itf].tx_ff, p_cdc->epin_buf, TU_ARRAY_SIZE(p_cdc->epin_buf));
@@ -169,6 +173,7 @@ uint32_t tud_cdc_n_write_flush (uint8_t itf)
   {
     TU_VERIFY( tud_cdc_n_connected(itf), 0 ); // fifo is empty if not connected
     TU_ASSERT( usbd_edpt_xfer(TUD_OPT_RHPORT, p_cdc->ep_in, p_cdc->epin_buf, count), 0 );
+    p_cdc->busy = true;
   }
 
   return count;
@@ -179,6 +184,15 @@ uint32_t tud_cdc_n_write_available (uint8_t itf)
   return tu_fifo_remaining(&_cdcd_itf[itf].tx_ff);
 }
 
+void tud_cdc_n_lock(uint8_t itf)
+{
+  osal_mutex_lock(_cdcd_itf[itf].mutex, OSAL_TIMEOUT_WAIT_FOREVER);
+}
+
+void tud_cdc_n_unlock(uint8_t itf)
+{
+  osal_mutex_unlock(_cdcd_itf[itf].mutex);
+}
 
 //--------------------------------------------------------------------+
 // USBD Driver API
@@ -206,6 +220,8 @@ void cdcd_init(void)
 #if CFG_FIFO_MUTEX
     tu_fifo_config_mutex(&p_cdc->rx_ff, osal_mutex_create(&p_cdc->rx_ff_mutex));
     tu_fifo_config_mutex(&p_cdc->tx_ff, osal_mutex_create(&p_cdc->tx_ff_mutex));
+    p_cdc->mutex = osal_mutex_create(&p_cdc->mutex_def);
+    p_cdc->busy = false;
 #endif
   }
 }
@@ -415,6 +431,8 @@ bool cdcd_xfer_cb(uint8_t rhport, uint8_t ep_addr, xfer_result_t result, uint32_
   //       Though maybe the baudrate is not really important !!!
   if ( ep_addr == p_cdc->ep_in )
   {
+    tud_cdc_n_lock(itf);
+    p_cdc->busy = false;
     if ( 0 == tud_cdc_n_write_flush(itf) )
     {
       // There is no data left, a ZLP should be sent if
@@ -424,6 +442,7 @@ bool cdcd_xfer_cb(uint8_t rhport, uint8_t ep_addr, xfer_result_t result, uint32_
         usbd_edpt_xfer(TUD_OPT_RHPORT, p_cdc->ep_in, NULL, 0);
       }
     }
+    tud_cdc_n_unlock(itf);
   }
 
   // nothing to do with notif endpoint for now
diff --git a/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/cdc/cdc_device.h b/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/cdc/cdc_device.h
index 7cef3fa..099ed06 100644
--- a/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/cdc/cdc_device.h
+++ b/cores/nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src/class/cdc/cdc_device.h
@@ -213,9 +213,15 @@ static inline uint32_t tud_cdc_write_str (char const* str)
   return tud_cdc_n_write_str(0, str);
 }
 
+void tud_cdc_n_lock(uint8_t itf);
+void tud_cdc_n_unlock(uint8_t itf);
 static inline uint32_t tud_cdc_write_flush (void)
 {
-  return tud_cdc_n_write_flush(0);
+  uint32_t ret;
+  tud_cdc_n_lock(0);
+  ret = tud_cdc_n_write_flush(0);
+  tud_cdc_n_unlock(0);
+  return ret;
 }
 
 static inline uint32_t tud_cdc_write_available(void)
```

hathach, 2020-09-02:
Hihi, thank for the patch file, though can you submit a pr for this. PR makes things easier to review the diff

arms22, 2020-09-03:
hi hathach .
I posted a pull request.
Please check it out.

hathach, 2020-09-04:
I am able to reproduce the data dropped, however, but it only happens when using the mutex as in your sketch. Without mutex in example sketch, the println() is perfect hmm :thinking: :thinking: I will take more closer look at the issue and the PR. Thanks for bringing this up. 

```
Bluefruit52 USB Serial Test
-----------------------------------

Aaaaaaaaaaaaaabbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
AaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaCcccBbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
AcccBbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
AaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaCcccBbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbaaaaaaaaaaaaaaaaaaaaaaaaaaaCcccBbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
AaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaCcccBbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
AaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaCcccBbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
Aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaabbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
AaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaCcccBbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
AaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaCcccBbbbbbbbbbbbbbbbbbaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaCcccBbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
Aaaaaaaaaaaaaaaaaaabbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
AaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaCcccBbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb
```

hathach, 2020-09-07:
@arms22 here is my break out on the issue https://github.com/hathach/tinyusb/issues/507 . I don't really like the proposed solution by https://github.com/adafruit/Adafruit_TinyUSB_ArduinoCore/pull/13 since the issue can be applied to other class with the same mechanism as well. I will try to find a better fix.

hathach, 2020-09-09:
@arms22 I make a  edpt level fix here, since this race pretty much applies to system wide, it is best to fix it at endpoint level else we will need to have each mutex for each class driver API that has this issue. 
https://github.com/hathach/tinyusb/pull/508

PR #577 (2020-09-15): update tinyusb core to commit 2afb2320e03e26c765aaeb51d430325b8adcdbf9
which contain the endpoint race fix from upstream. fix #565
closed 2020-09-15 by hathach