adafruit/Adafruit_nRF52_Arduino #566: BLEConnection::requestPairing does not return if the device is disconnected during pairing.

arms22, 2020-09-03:
**Describe the bug**
`BLEConnection::requestPairing` does not return if the device is disconnected during pairing.

After the `BLE_GAP_EVT_DISCONNECTED` event occurs, the `BLE_GAP_EVT_EVT_CONN_SEC_UPDATE` event does not occur, so
`xSemaphoreGive(_pair_sem)`is not called and No longer return from `xSemaphoreTake(_pair_sem, portMAX_DELAY).`

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.13 Windows 10
 - **BSP**   : 0.20.5
 - **Board** : Feather nRF52840 Express
 - **Sketch**: None

hathach, 2020-09-09:
ah right, thanks, that is indeed an issue. Currently there is huge pending PR $466 that implement LESC and completely rework the pairing process. We will wait until that PR got merged and double check this again. Thanks for posting.

arms22, 2020-09-10:
Hi hathach
Thanks for the reply.
I'll check again once  PR $466 have been merged!

hathach, 2020-09-11:
#466 is merged, I have just checked the log. It completely reworks the pairing process from using legacy just work to more advanced LESC. Some API is removed and new is added. The pairing process is not non-blocking, a callback is invoked when it is complete (failed or succeeded). Thanks for filing the issue.

closed 2020-09-11 by hathach