adafruit/Adafruit_nRF52_Arduino #573: [Feather nRF52840 Express] Using Serial1 : undefined reference to `vtable for HardwareSerial'

hugoschaaf98, 2020-09-09:
Hello, I'm experiencing the following issue. The link edition fails with the following error 
``
C:\Users\Me\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.21.0\cores\nRF5/HardwareSerial.h:61: undefined reference to `vtable for HardwareSerial'
``

collect2.exe: error: ld returned 1 exit status
Utilisation de la bibliothèque Wire version 1.0 dans le dossier: C:\Users\hschaaf\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.21.0\libraries\Wire 
exit status 1

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.12 on Windows 10
 - **BSP**   : 0.21.0
 - **Board** : Feather nRF52840 Express
 - **Sketch**: /Bluefruit52Lib/examples/Hardware/Serial1_test/Serial1_test.ino

I don't understand why since Serial1 is declared in Uart.h and defined in Uart.cpp ...

Regards, 
Hugo

ladyada, 2020-09-09:
try installing 1.8.13 in case it makes a difference

hugoschaaf98, 2020-09-09:
I'm sorry but I'm still getting the error with Arduino 1.8.13.

`Linking everything together...
"C:\\Users\\hschaaf\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-gcc" "-LC:\\Users\\hschaaf\\AppData\\Local\\Temp\\arduino_build_269180" -O0 -Wl,--gc-sections -save-temps "-LC:\\Users\\hschaaf\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.21.0\\cores\\nRF5/linker" -Tnrf52840_s140_v6.ld "-Wl,-Map,C:\\Users\\hschaaf\\AppData\\Local\\Temp\\arduino_build_269180/Serial1_test.ino.map" -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align --specs=nano.specs --specs=nosys.specs -o "C:\\Users\\hschaaf\\AppData\\Local\\Temp\\arduino_build_269180/Serial1_test.ino.elf" "C:\\Users\\hschaaf\\AppData\\Local\\Temp\\arduino_build_269180\\sketch\\Serial1_test.ino.cpp.o" "C:\\Users\\hschaaf\\AppData\\Local\\Temp\\arduino_build_269180\\core\\variant.cpp.o" -Wl,--start-group -lm "C:\\Users\\hschaaf\\AppData\\Local\\Temp\\arduino_build_269180/core\\core.a" -Wl,--end-group
c:/users/hschaaf/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\hschaaf\AppData\Local\Temp\arduino_build_269180/core\core.a(Uart.cpp.o): in function `HardwareSerial::HardwareSerial()':
C:\Users\hschaaf\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.21.0\cores\nRF5/HardwareSerial.h:61: undefined reference to `vtable for HardwareSerial'
collect2.exe: error: ld returned 1 exit status
exit status 1
Erreur de compilation pour la carte Adafruit Feather nRF52840 Express
`

hugoschaaf98, 2020-09-09:
I'm sorry but I'm still getting the error with Arduino 1.8.13.

``
Linking everything together...
"C:\\Users\\hschaaf\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-gcc" "-LC:\\Users\\hschaaf\\AppData\\Local\\Temp\\arduino_build_269180" -O0 -Wl,--gc-sections -save-temps "-LC:\\Users\\hschaaf\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.21.0\\cores\\nRF5/linker" -Tnrf52840_s140_v6.ld "-Wl,-Map,C:\\Users\\hschaaf\\AppData\\Local\\Temp\\arduino_build_269180/Serial1_test.ino.map" -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align --specs=nano.specs --specs=nosys.specs -o "C:\\Users\\hschaaf\\AppData\\Local\\Temp\\arduino_build_269180/Serial1_test.ino.elf" "C:\\Users\\hschaaf\\AppData\\Local\\Temp\\arduino_build_269180\\sketch\\Serial1_test.ino.cpp.o" "C:\\Users\\hschaaf\\AppData\\Local\\Temp\\arduino_build_269180\\core\\variant.cpp.o" -Wl,--start-group -lm "C:\\Users\\hschaaf\\AppData\\Local\\Temp\\arduino_build_269180/core\\core.a" -Wl,--end-group
c:/users/hschaaf/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\hschaaf\AppData\Local\Temp\arduino_build_269180/core\core.a(Uart.cpp.o): in function `HardwareSerial::HardwareSerial()':
C:\Users\hschaaf\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.21.0\cores\nRF5/HardwareSerial.h:61: undefined reference to `vtable for HardwareSerial'
collect2.exe: error: ld returned 1 exit status
exit status 1
Erreur de compilation pour la carte Adafruit Feather nRF52840 Express
``

hathach, 2020-09-11:
If you installed Arduino via windows store, please remove and try to download and install it directly from arduino.cc to see if that makes any differences.

hathach, 2020-09-14:
@hugoschaaf98 said his connectivity and/or browser is causing the duplicated post unintentionally. Please be mindful with the same situation since this does indeed generated tons of email notification toward maintainers.

hugoschaaf98, 2020-09-14:
Thank you for unlocking the conversation. I'm sincerely sorry for the inconvenience...

Upgrading Arduino didn't solve the issue. I totally forgot to notice that I modified the compiler's optimization options !
I changed -Ofast to -O0 in order to debug properly. And this is precisely that modification which causes the linking to fail.

hathach, 2020-09-14:
did you changed all the -Ofast to -O0, recent update make it easier to change the optimization. If you haven't updated it yet, please give it a try
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/platform.txt#L32

hugoschaaf98, 2020-09-14:
This is precisely what I've done...

hathach, 2020-09-14:
Hmm, nothing top of my head could explain it. Maybe it miss something. Right now I don't have time but will look at this later on

hugoschaaf98, 2020-09-14:
If I find something I'll let you know. This linking error seems to be common with c++ and I would suspect something with the virtual constructors...

hathach, 2020-09-14:
if you only need to work with debug, you could also try with `-Og` which is optimize for debug to see if that works

hathach, 2021-06-03:
closed due to lack of activities and it is not reproducible on BSP package release

closed 2021-06-03 by hathach