adafruit/Adafruit_nRF52_Arduino #580: Bluefruit52Lib/examples/Peripheral/beacon on Feather nRF52840 Express doesn't work

mcecchi, 2020-09-24:
**Describe the bug**
suspendLoop() block any Serial.println. Furthermore, both "nRF Beacon" and "nRF connect" apps by Nordic cannot find device.

Commenting out suspendLoop()  Serial.println back to work, "nRF connect" sees beacon correctly, but "nRF Beacon" gets stuck on "Scanning".

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.13 on Windows
 - **BSP**   : 0.21.0
 - **Board** : Feather nRF52840 Express
 - **Sketch**: Bluefruit52Lib/examples/Peripheral/beacon

**To Reproduce**
Compile, flash and run Arduino Bluefruit52Lib/examples/Peripheral/beacon on Feather nRF52840 Express.

hathach, 2020-10-05:
hi, I just tested with latest code in github, it is probably due to the blocking wait for the Serial connection. Would you mind commenting it out and try again. We will also comment that out to prevent user confusion.

mcecchi, 2020-10-05:
Hi @hathach thanks for your quicly response.

Commenting blocking wait doesn't resolve. The only way I found is commenting out suspendLoop() and leaving blocking wait in place!

hathach, 2020-10-05:
I couldn't reproduce it, please try with the current github tip of master.

variistechnologies, 2020-12-05:
Same issue here with the code from master, nRF Beacon app not finding any beacons, stuck on scanning.

variistechnologies, 2020-12-05:
Actually I think nRF Beacon app is broken (apparently not supported anymore), I'm able to see my beacon using another 3rd party app.

hathach, 2021-06-03:
just tested, and nrfconnect can detect the beacon data. Note: the nrf beacon on iOS seems to be obsolete and abandoned. You should use the nrfconnect instead. #655 also change default manufacture to Nordic and match uudi and major, minor to value used by beacon example in nordic sdk.

PR #655 (2021-06-03): update beacon example
fix #580 
- change default manuf to Nordic
- use same uuid, major, minor as nordic beacon example
closed 2021-06-03 by hathach