adafruit/Adafruit_nRF52_Arduino #581: Error 6: Operation failed when doing OTA in iOS

glucee, 2020-09-25:
**Describe the bug**
When operating OTA update in iOS, we found an issue like this:
xx means the ID

peripheral.writeValue(0x0201, for: xx, type: .withResponse)
Data written to xx
Notification received from xx value (0x): 100206
Response (Op Code = 2, Status = 6) received
Error 6: Operation failed
Writing to characteristic xx...
peripheral.writeValue(0x06, for: xx, type: .withResponse)
Notification received from xx, value (0x): 100206
Response (Op Code = 2, Status = 6) received

**Set up (mandatory)**
 - **PC & IDE** : Windows 10
 - **BSP**   : 0.20.0
 - **Board** : Feather nRF52840 Express

hathach, 2020-09-25:
Please provide detail on how to reproduce the issue, sketch to run on nRF, app to run on iOS and exact steps to cause the issue.

glucee, 2020-09-25:
> Please provide detail on how to reproduce the issue, sketch to run on nRF, app to run on iOS and exact steps to cause the issue.

@hathach sure, have updated

hathach, 2020-09-25:
thanks, we haven't done much testing with OTA for quite some time. I will test this out when I have time.

supersunil, 2020-12-28:
Exactly the same problem I'm facing.

I've flashed the bootloader and programmed my Adafruit nRF52840 Express board to perform the OTA DFU operation.

I've opened the example code and compiled it using Arduino 1.8.13 and I've double checked the bootloader version (0.3.2 with softdevice s140 6.1.1) on the board as well as in Arduino IDE.

I've moved the compiled code .ino.zip to my Android Device running Android 8.0.0. I've installed the nRF Connect 4.24.3 and followed the process of OTA Device Firmware Update, it was successful (shown as video in the attached drive link)

https://drive.google.com/file/d/1wC63r85xd41FOQqn4Ciw4HilG8Ex8iqu/view?usp=sharing

Now I'm performing the exact same process on my iPhone running iOS 14.3 and the nRF Connect 2.4.6 but it exits with operation failed status (shown as video in attached drive link)

https://drive.google.com/file/d/1UuRIBfP-XKB9VHR4V0gn2aGhyWrbDUoN/view?usp=sharing

Help me to get rid of it.

aovestdipaperino, 2023-01-18:
Any progress on this?