adafruit/Adafruit_nRF52_Arduino #585: Add function to remap BLE Connection Status LED (Blue LED) to any other Pin/GPIO

jpconstantineau, 2020-09-30:
**Is your feature request related to a problem? Please describe.**
I want to use another pin/gpio to show status of the connection so that I can use another, external LED to show connection status.

**Describe the solution you'd like**
Add a function called "Bluefruit.changeConnLed(PIN)" to make it possible to change from the default BLUE_LED to anything the user wants. This will require setting up a static private variable that set to LED_BLUE by default, use the new variable instead of the macro for all LED operations and adding this function to change the status GPIO.

**Describe alternatives you've considered**
Manually setting the logic with the connect and disconnect callbacks essentially duplicates code and does not re-use the blink timer on "not connected" status.

**Additional context**
This feature would make it possible to have the BLE status shown out when the feather is enclosed in a case.
This feature would also make it possible for 3rd party boards to re-use the nrf52832 and PCA10056 definitions where their status LEDs cannot use the default ones.

I am making this feature request in preparation for me to implement it in the BSP.  I want your comments on the proposed implementation above prior to implementation and submitting a PR.

hathach, 2020-10-01:
you should use the adv stop, connect callback etc for setting custom indicator. The reason is I have a plan to remove the usage of built-in connection LED. It is nice with first release, but with dual role (prph + central) and/or concurrent connections. Indicator becomes too complex to maintain consistent. It should be managed by application for its need though.

closed 2020-10-01 by hathach

jpconstantineau, 2020-10-01:
Thanks.  Good thing I asked before starting work ahead of a PR.  Are you planning to have helper functions tied to the LED or completely removing it from the library?

I agree that with multiple connections, managing a single LED to convey different states is not trivial.