adafruit/Adafruit_nRF52_Arduino #590: Feature - NRF52810 support

wb8wka, 2020-10-07:
This is both a question and feature request.  

**Summary**  

The nrf52510 also can use the S132 softdevice.  Is there a easy way to drop in existing Adafruit arduino code for the NRF52832 into this device? I imagine a similar question would exist for the bootloader as well.

**Is your feature request related to a problem? Please describe.**

Hardware cost. The 510 is a lower cost chip that may work in some scenarios.

**Describe the solution you'd like**

Generic NRF52810 support, with perhaps some configurable board settings

**Describe alternatives you've considered**

Segger embedded studio - very convoluted compared to the the Adafruit Arduino package

**Additional context**

 Is there a easy way to drop in existing Adafruit arduino code for the NRF52832 into this device? I imagine a similar question would exist for the bootloader as well.

hathach, 2020-10-07:
we have no plan doing this, but any PRs would be welcome. 

> Is there a easy way to drop in existing Adafruit arduino code for the NRF52832 into this device? I imagine a similar question would exist for the bootloader as well.

There is no easy way, you have to port all the way from bottom to top of the BSP.

wb8wka, 2020-10-07:
Thank you, that is what I figured.   Doing some dev on SES and I will say, your Arduino implementation is so much easier.  What takes minutes is hours of digging on SES, at least for this newbie.  

Close this one out

closed 2020-10-07 by wb8wka

Erickrk, 2020-10-29:
Hey @wb8wka think I've seen what you're looking for. Look at this guy right here: [https://github.com/sandeepmistry/arduino-nRF5](url)

Wish you all the luck.

wb8wka, 2020-10-30:
Thank you. I knew of this but had forgotten about it. It looks like it may
have gone dormant but I'll still give it a try.  This likely(?) is less of
an issue since the softdevice isn't tied to the bootloader and you make the
calls directly from reading the description.  I'll post back once my
hardware arrives next week