adafruit/Adafruit_nRF52_Arduino #593: isWhiteSpace should match \r, \n, \f

shurik179, 2020-10-10:
According to docs on arduino website: https://www.arduino.cc/reference/tr/language/functions/characters/iswhitespace/ and common conventions, isWhiteSpace(c) should match space, \t, \v,  \n, \r, \f . 

Yet the implementation of this function  in Adafruit_nRF52_Arduino/cores/nRF5/WCharacter.h only matches space and \t

ladyada, 2020-10-10:
thanks could you submit a PR? :)

shurik179, 2020-10-11:
Actually I checked other cores and found that `isWhitespace` and `isSpace` in all of them is consistent with each other: `isWhitespace` is same as standard C `isblank` and `isSpace` is same as C `isspace`. Thus, it makes no sense to change implementation in Adafruit core. Instead, what really needs to be changed is documentation on Arduino website. 

I am closing this issue then - sorry for the noise - and will file a bug report on Arduino website.

closed 2020-10-11 by shurik179