adafruit/Adafruit_nRF52_Arduino #595: Github Actions fails with latest Python 3.9

jpconstantineau, 2020-10-22:
**Describe the bug**
Github actions now uses Python 3.9 to run the build scripts.  The build scripts fails with plenty of errors.
Most common error: AttributeError: 'array.array' object has no attribute 'tostring'

short term: Recommended action: force github actions to use prior version of python.
long term: update the scripts.

**Set up (mandatory)**
 github actions enabled and running as per configured actions in yaml file.

**To Reproduce**
Do a git push to run the action scripts.

**Screenshots**
Check example run from a clone of the repo:
https://github.com/jpconstantineau/Adafruit_nRF52_Arduino/runs/1295170248?check_suite_focus=true

ladyada, 2020-10-23:
thanks, can you submit a PR that changes the python version to 3.8 instead of 3.x? taht should help fix

hathach, 2020-11-02:
adafruit-nrfutil new release `post13` fixed the compatible issue with 3.9. Therefore we don't have to force the python version. Please try to update it with `pip3 install adafruit-nrftuil` . Thank you very much for the issue and pr.

closed 2020-11-02 by hathach