adafruit/Adafruit_nRF52_Arduino #602: StandardFirmataBLE.ino BUG: friend declaration of 'void encodeByteStream(size_t, uint8_t*, size_t) const'

Achillesy, 2020-11-12:
**Source Code**
https://raw.githubusercontent.com/adafruit/Adafruit_nRF52_Arduino/master/libraries/Bluefruit52Lib/examples/Peripheral/StandardFirmataBLE/StandardFirmataBLE.ino

**Serial Log**
Arduino: 1.8.13 (Windows 10), TD: 1.53, Board: "Adafruit Feather nRF52832, 0.3.2 SoftDevice s132 6.1.1, Level 0 (Release)"

In file included from C:\Workspace\Arduino\Adafruit_nRF52\Feather_nRF52832\StandardFirmataBLE\StandardFirmataBLE.ino:31:

C:\Program Files (x86)\Arduino\libraries\Firmata/Firmata.h:131:17: error: friend declaration of 'void encodeByteStream(size_t, uint8_t*, size_t) const' specifies default arguments and isn't a definition [-fpermissive]

  131 |     friend void FirmataMarshaller::encodeByteStream (size_t bytec, uint8_t * bytev, size_t max_bytes = 0) const;

Multiple libraries were found for "Servo.h"

 Used: C:\Workspace\Arduino\libraries\Servo

 Not used: C:\Program Files (x86)\Arduino\libraries\Servo

 Not used: C:\Users\Achilles\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.21.0\libraries\Servo

exit status 1

Error compiling for board Adafruit Feather nRF52832.

hathach, 2020-11-12:
what is your firmware library version, also make sure you use the Servo in the BSP `C:\Users\Achilles\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.21.0\libraries\Servo` instead of the one in the `C:\Workspace\Arduino\libraries\Servo`

Achillesy, 2020-11-12:
Thank you for your reply. 

Arduino: 1.8.13 (Windows 10), TD: 1.53, 
Board: "Adafruit Feather nRF52832, 0.3.2 SoftDevice s132 6.1.1, Level 0 (Release)"

**Servo**: by Michael Margolis, Arduino Version 1.1.7
**Adafruit BLEFirmata**: by Adafruit Version 1.1.2

Before reporting this bug, I searched for a solution, but it didn't work. 

After deleting "C:\Workspace\Arduino\libraries\Servo", 
Arduino will notify me to update automatically because this is its default library. 

How to use the specified library:  "C:\Users\Achilles\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.21.0\libraries\Servo" ?

Gymnae, 2020-11-13:
I am facing the same problem

hathach, 2020-11-14:
Can either of you provide your **firmata** library version, it should be part of the compile log. For the Servo, just remove the one installed in Arduino/libraries, it will then use the one come with the BSP. we will try to make change (e.g bump up version) later on to make sure it will pick the bsp packed one.

Gymnae, 2020-11-14:
My Arduino IDE version is: 
Firmata version: 2.5.8
I tried 2.5.7 same problem.
I tried both the Board Manager release and the git version of the BSP from this repo.
I deleted the Servo libraries not installed with this BSP:
```
sudo rm -rf /Applications/Arduino.app/Contents/Java/libraries/Servo
 sudo rm -rf /Users/<username>/Documents/Arduino/libraries/Servo
```
 
Here is the full print out from the compile command after:  
```
WARNING: library Servo claims to run on nrf52 architecture(s) and may be incompatible with your current board which runs on Adafruit_nRF52_Arduino architecture(s).
In file included from /Users/gymnae/Documents/Arduino/hardware/Adafruit/Adafruit_nRF52_Arduino/libraries/Bluefruit52Lib/examples/Peripheral/StandardFirmataBLE/StandardFirmataBLE.ino:31:
/Applications/Arduino.app/Contents/Java/libraries/Firmata/Firmata.h:131:17: error: friend declaration of 'void encodeByteStream(size_t, uint8_t*, size_t) const' specifies default arguments and isn't a definition [-fpermissive]
  131 |     friend void FirmataMarshaller::encodeByteStream (size_t bytec, uint8_t * bytev, size_t max_bytes = 0) const;
      |                 ^~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Adafruit Feather nRF52840 Express.

```

Achillesy, 2020-11-14:
Arduino: 1.8.13 (Windows 10), TD: 1.53,
Board: "Adafruit Feather nRF52832, 0.3.2 SoftDevice s132 6.1.1, Level 0 (Release)"

Servo: by Michael Margolis, Arduino Version 1.1.7
Adafruit BLEFirmata: by Adafruit Version 1.1.2

Model Number
Bluefruit Feather52
Firmware Revision
s132 6.1.1
Software Revision
0.21.0

As I said before, deleting the library Servo is useless.  : (

Achillesy, 2020-11-15:
Hi Gymnae,
open libraries/Firmata/Firmata.h:131
change: friend void FirmataMarshaller::encodeByteStream (size_t bytec, uint8_t * bytev, size_t max_bytes = 0) const;
        to: friend void FirmataMarshaller::encodeByteStream (size_t bytec, uint8_t * bytev, size_t max_bytes) const;

replace: C:\Workspace\Arduino\libraries\Servo
     with: C:\Users\Achilles\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.21.0\libraries\Servo

**It's working!**
Sketch uses 81460 bytes (28%) of program storage space. Maximum is 290816 bytes.
Global variables use 9936 bytes (19%) of dynamic memory, leaving 42288 bytes for local variables. Maximum is 52224 bytes.
Upgrading target on COM3 with DFU package C:\Temp\arduino_build_646705\StandardFirmataBLE.ino.zip. Flow control is disabled, Single bank, Touch disabled
########################################
########################################
########################################
########################################
Activating new firmware
Device programmed.

closed 2020-11-15 by Achillesy

Gymnae, 2020-11-15:
Maybe keep this bug open because your fix is not upstream? Or would this bug with the library belong to https://github.com/firmata/arduino?

Achillesy, 2020-11-15:
Actually, these are two bugs:

The First bug is in Firmata.h: the declaration of encodeByteStream should not include default value (=0)

The Second bug is Multiple libraries "Servo.h"

They do not belong to this script,  so I closed this issue.  : )

hathach, 2020-11-16:
It is indeed issue with firmata library, it is fixed in their upstream https://github.com/firmata/arduino/blob/master/Firmata.h#L131 here. But it is not released just yet, I have no control when they release `2.5.9` , maybe you could just download their current master.zip and extract to the libraries folder.

@Achillesy you don't need the `Adafruit BLEFirmata` library, that one is for nrf51 series I think.

Achillesy, 2020-11-16:
@hathach Thanks.

therealhugocosta, 2022-05-02:
I had this issue, as per @hathach suggestion i replaced the Firmata folder in libraries folder with the current master.
Now it works. Thank you!