adafruit/Adafruit_nRF52_Arduino #603: Firmata sketch no control via serial port via BLE

Gymnae, 2020-11-18:
**Describe the bug**
When trying to use a pseudo-tty to control the Feather via BLE, a timeout occurs. Directly connecting to the targeted device, in my case an Intel NUC running CentOS and node-RED also doesn't work when trying to control it via `/dev/ttyACM0`. 

But when I flash the stock StandardFirmata sketch, I can control the device via `/dev/ttyACM0`. To my newbie eyes it seems that the serial communication does not work properly between the Feather and the pseudo-tty created by ble-serial.
I tried adjusting the Baud rate to 57600 in the sketch, as I read this solved the problem for another person: https://github.com/adafruit/Adafruit_BLE_PinIO/issues/10

Only control via the Adafruit Connect iOS app is possible with the StandardFirmataBLE sketch.

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.13 macOS 10.15
 - **BSP**   : 0.21.0
 - **Board** : Feather nRF52840 Express
 - **Sketch**: Bluefruit52Lib/examples/Peripheral/StandardFirmataBLE

**To Reproduce**
Steps to reproduce the behavior:
1. Flash StandardFirmataBLE onto Feather nRF52840 Express
2. Install [node-red-contrib-johnny5](https://flows.nodered.org/node/node-red-contrib-johnny5) as node in node-RED (and ensure that python/node have permissions to bluetooth hardware)
3. Install [ble-seria](https://pypi.org/project/ble-serial/)l and connect to the Feather: `ble-serial -v -d <mac_address> -t random  -w 6e400002-b5a3-f393-e0a9-e50e24dcca9e`
3. Setup a new Firmata node and add /dev/pts/2 as serial device
4. Try to connect, see timeouts

**Serial Log**

This is the output of ble-serial, which does connect to the feather, as also visible by a permanent blue LED
```
ble-serial -v -d <my_mac_address> -t random  -w 6e400002-b5a3-f393-e0a9-e50e24dcca9e
18:36:43.670 | INFO | virtual_serial.py: Slave created on /tmp/ttyBLE -> /dev/pts/1
18:36:43.943 | INFO | interface.py: Connected device <my_mac_address>
18:36:44.243 | DEBUG | interface.py: Found write characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e
18:36:44.243 | INFO | interface.py: Receiver set up
18:36:44.243 | INFO | __main__.py: Running main loop!
18:36:49.347 | DEBUG | virtual_serial.py: Read: b'\xf9'
18:36:49.347 | DEBUG | interface.py: Sending b'\xf9'
18:36:49.349 | DEBUG | virtual_serial.py: Read: b'\xf0y\xf7'
18:36:49.349 | DEBUG | interface.py: Sending b'\xf0y\xf7'
18:36:54.367 | DEBUG | virtual_serial.py: Read: b'\xf9'
18:36:54.367 | DEBUG | interface.py: Sending b'\xf9'
18:36:54.375 | DEBUG | virtual_serial.py: Read: b'\xf0y\xf7'
18:36:54.375 | DEBUG | interface.py: Sending b'\xf0y\xf7'
```

hathach, 2020-11-19:
It is not clear to me this a bug of this repo, it could be bug of your ble-serial, and/or how to use it. Which we don't know enough to support.

closed 2020-11-19 by hathach

Gymnae, 2020-11-19:
It might be the combination of ble-serial and adafruit's FirmataStandardBLE sketch.

Is the FirmataStandardBLE sketch supposed to work like a FirmataStandard when connected locally and addressable as an ACM USB modem via ttyACM*? 

FirmataStandard Firmata-Express work as desired when I address them via ttyACM*,  but your FirmataStandardBLE won't work with python code or local test programs issuing Firmata commands (like blink an LED) when connected locally via USB. 

Is therefore your FirmataStandardBLE designed to be only used with Blufruit connect? Or could you tell me if there's a special way Blufruit connect connects to Blufruit devices, which enables it to send commands like Pin I/O?

hathach, 2020-11-19:
In short, no. ble uart (nus) is not the same as wire uart. It depend on many parameter of the gatt protocol and on way central send/receive message as well.Though If you work on this further enough, maybe you could figure why it doesn't work and can tune-up ble parameter, configure etc... but it is not what we could support.

Gymnae, 2020-11-19:
Ah, ok. Thank you for the info and explaining, I appreciate you taking the time.
Then I will stop on this path to find a different path to control the PINs without bluefruit connect remotely.