adafruit/Adafruit_nRF52_Arduino #606: Can't include ArduinoBLE.h and Arduino_LSM9DS1.h together and other problems

chinkapinoak, 2020-12-07:
See problems description and code below.
/*
  BluetoothClient - BLE client trial code
  December 6, 2020
  Problems:
    BLE device will not be recognized by LightBlue unless 
    1) There is an analogRead() function called.
    2) Globals are not used. A while(1) within Loop let's one use the locals since globals can't be used.
    4) static variables are not used.
    5) A third characteristic is defined (csEnabled in this case). Only two characteristics are added to the service.
    6) Arduino_LSM9DS1.h is NOT included. Thus one can't report gyro and accelerometer data via Bluetooth.
  Using Arduino IDE version 1.8.13 (latest as of Dec 6, 2020
*/
#include <ArduinoBLE.h>
//?? #include <Arduino_LSM9DS1.h>

// create service:
BLEService countService("10000000");
// create service characteristics.
BLEUnsignedLongCharacteristic csSeconds("10000001", BLERead);
BLEUnsignedLongCharacteristic csCount("10000002", BLERead | BLENotify);
  // This one is declared but not added to the service.
BLEIntCharacteristic csEnabled("19B10010-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite );

void setup() {
  Serial.begin(9600);
  pinMode(LED_BUILTIN, OUTPUT);

  // begin initialization
  while (!BLE.begin()) {
    Serial.println("Waiting for BLE to start");
    delay(1);
  }

  // set the local name that the peripheral advertises:
  BLE.setLocalName("Count Service");
  // set the UUID for the service:
  BLE.setAdvertisedService(countService);

  // add the characteristics to the service
  countService.addCharacteristic(csSeconds);
  countService.addCharacteristic(csCount);
  // add the service
  BLE.addService(countService);

  // start advertising the service:
  BLE.advertise();
}

void loop() {
  unsigned long lastSec=millis()/1000; // Coded such that lastSec will go from 0..2**32-1
  unsigned long loopCount=0;
  uint16_t enabled=0;

  while(1) {  // Use while to get around global varialbe usage when using arduinoBLE.h
    if (lastSec != millis()/1000) lastSec++;

    BLEDevice central = BLE.central();
    // if a central is connected to the peripheral:
    if (central) {
      // print the central's BT address:
      Serial.print("Connected to central: ");
      Serial.println(central.address());
      // turn on LED to indicate connection:
      digitalWrite(LED_BUILTIN, HIGH);
    
      // while the central remains connected:
      while (central.connected()) {
        loopCount = loopCount + 1;
        //************** DON'T remove this analogRead. Device will not be recognized otherwise.
        int xReading = analogRead(A0);
        //**************
        // write values to service characteristics:
        if (lastSec != millis()/1000) lastSec++;
        csSeconds.writeValue(lastSec);
        csCount.writeValue(loopCount);
      }
      loopCount +=1000000;
      Serial.println("Disconnected from central.");
    } else {
      // turn off the LED
      digitalWrite(LED_BUILTIN, LOW);
    }
  }
}

hathach, 2020-12-07:
please fill in all the required fields in the bug issue template, and put all your code in the block code.

chinkapinoak, 2020-12-07:
**Describe the bug**
Using the ArduinoBLE library has limitations in getting the Nano BLE to be recognized by LightBlue on an Android phone.
BLE device will not be recognized by LightBlue unless
1) There is an analogRead() function called in the program
2) Globals are not used. A while(1) within Loop let's one use the locals since globals can't be used.
3) Only three characteristics are allowed.
4) When trying to use two characteristics, a third characteristic must be defined (csEnabled in this case). Only two characteristics are added to the service.
5) Can't include "Arduino_LSM9DS1.h". Thus one can't report gyro and accelerometer data via Bluetooth.

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.13 on Windows 10
 - **BSP**   : 0.20.0 ??
 - **Board** : Arduino Nano 33 BLE

 - **Sketch**: [NanoBleCompileProblems.ino.docx](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/5653303/NanoBleCompileProblems.ino.docx)

**To Reproduce**
Steps to reproduce the behavior:
1. Load the  compiled NanoBleCompileProblems.ino into a Nano 33 BLE
2. Monitor with LightBlue on an Android. "Count Service" should be seen in bold letters and can be connected to with two characteristics.
3. Comment out the AnalogRead(A0) line for problem 1 replication.
4. Reload. On Android "Count Service" is seen but not in bold  letters meaning it's not seen.
5. Uncomment the AnalogRead(A0) line.
6. Add a global declaration below the #include lines. Like "int dummy;". This is for problem 2 replication.
7. Reload. On Android "Count Service" is seen but not in bold  letters meaning it's not seen.
8. Delete the "int dummy;" line.
9. Comment out the "BLEIntCharacteristic csEnabled" line to replicate problem 4.
10. Reload. On Android "Count Service" is seen but not in bold  letters meaning it's not seen.
11. Uncomment the "#include <Arduino_LSM9DS1.h>" line to replicated problem 5.
12. Reload. On Android "Count Service" is seen but not in bold  letters meaning it's not seen.
13. Comment out the "#include <Arduino_LSM9DS1.h>" line.
14. Reload. The "Count Service" on the Android will show bold meaning the Nano BLE is now recognized and can to connected to.

**Screenshots**
None

**Serial Log**
None
``

`/*
  BluetoothClient - BLE client trial code
  Phil Bird - December 6, 2020
  Problems:
    BLE devices will not be recognized by LightBlue unless 
    1) There is an analogRead() function called.
    2) Globals are not used. A while(1) within Loop let's one use the locals since globals can't be used.
    4) static variables are not used.
    5) A third characteristic is defined (csEnabled in this case). Only two characteristics are added to the service.
  Using Arduino IDE version 1.8.13 (latest as of Dec 6, 2020
*/
#include <ArduinoBLE.h>
//?? #include <Arduino_LSM9DS1.h>

// create service:
BLEService countService("10000000");
// create service characteristics.
BLEUnsignedLongCharacteristic csSeconds("10000001", BLERead);
BLEUnsignedLongCharacteristic csCount("10000002", BLERead | BLENotify);
  // This one is declared but not added to the service.
BLEIntCharacteristic csEnabled("19B10010-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite );

void setup() {
  Serial.begin(9600);
  pinMode(LED_BUILTIN, OUTPUT);

  // begin initialization
  while (!BLE.begin()) {
    Serial.println("Waiting for BLE to start");
    delay(1);
  }

  // set the local name that the peripheral advertises:
  BLE.setLocalName("Count Service");
  // set the UUID for the service:
  BLE.setAdvertisedService(countService);

  // add the characteristics to the service
  countService.addCharacteristic(csSeconds);
  countService.addCharacteristic(csCount);
  // add the service
  BLE.addService(countService);

  // start advertising the service:
  BLE.advertise();
}

void loop() {
  unsigned long lastSec=millis()/1000; // Coded such that lastSec will go from 0..2**32-1
  unsigned long loopCount=0;
  uint16_t enabled=0;

  while(1) {  // Use while to get around global varialbe usage when using arduinoBLE.h
    if (lastSec != millis()/1000) lastSec++;

    BLEDevice central = BLE.central();
    // if a central is connected to the peripheral:
    if (central) {
      // print the central's BT address:
      Serial.print("Connected to central: ");
      Serial.println(central.address());
      // turn on LED to indicate connection:
      digitalWrite(LED_BUILTIN, HIGH);
    
      // while the central remains connected:
      while (central.connected()) {
        loopCount = loopCount + 1;
        //************** DON'T remove this analogRead. Device will not be recognized otherwise.
        int xReading = analogRead(A0);
        //**************
        // write values to service characteristics:
        if (lastSec != millis()/1000) lastSec++;
        csSeconds.writeValue(lastSec);
        csCount.writeValue(loopCount);
      }
      loopCount +=1000000;
      Serial.println("Disconnected from central.");
    } else {
      // turn off the LED
      digitalWrite(LED_BUILTIN, LOW);
    }
  }
}

ladyada, 2020-12-07:
you're using Arduino BLE not Adafruit - please open this issue with Arduino's library
thanx :)

closed 2020-12-07 by ladyada

chinkapinoak, 2020-12-07:
Hopefully I've now provided necessary information. If not let me know how to get it right.Thanks. Phil Bird

darkworks, 2023-09-25:
how to achieve something like this with adafruit lib instead of Arduino BLE lib. where we can create service define its characteristics and send and receive data over it
```
*/
#include <ArduinoBLE.h>
//?? #include <Arduino_LSM9DS1.h>

// create service:
BLEService countService("10000000");
// create service characteristics.
BLEUnsignedLongCharacteristic csSeconds("10000001", BLERead);
BLEUnsignedLongCharacteristic csCount("10000002", BLERead | BLENotify);
// This one is declared but not added to the service.
BLEIntCharacteristic csEnabled("19B10010-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite );

void setup() {
Serial.begin(9600);
pinMode(LED_BUILTIN, OUTPUT);

// begin initialization
while (!BLE.begin()) {
Serial.println("Waiting for BLE to start");
delay(1);
}

// set the local name that the peripheral advertises:
BLE.setLocalName("Count Service");
// set the UUID for the service:
BLE.setAdvertisedService(countService);

// add the characteristics to the service
countService.addCharacteristic(csSeconds);
countService.addCharacteristic(csCount);
// add the service
BLE.addService(countService);

// start advertising the service:
BLE.advertise();
}

void loop() {
unsigned long lastSec=millis()/1000; // Coded such that lastSec will go from 0..2**32-1
unsigned long loopCount=0;
uint16_t enabled=0;

while(1) { // Use while to get around global varialbe usage when using arduinoBLE.h
if (lastSec != millis()/1000) lastSec++;

BLEDevice central = BLE.central();
// if a central is connected to the peripheral:
if (central) {
  // print the central's BT address:
  Serial.print("Connected to central: ");
  Serial.println(central.address());
  // turn on LED to indicate connection:
  digitalWrite(LED_BUILTIN, HIGH);

  // while the central remains connected:
  while (central.connected()) {
    loopCount = loopCount + 1;
    //************** DON'T remove this analogRead. Device will not be recognized otherwise.
    int xReading = analogRead(A0);
    //**************
    // write values to service characteristics:
    if (lastSec != millis()/1000) lastSec++;
    csSeconds.writeValue(lastSec);
    csCount.writeValue(loopCount);
  }
  loopCount +=1000000;
  Serial.println("Disconnected from central.");
} else {
  // turn off the LED
  digitalWrite(LED_BUILTIN, LOW);
}
}
}
```

thanks