adafruit/Adafruit_nRF52_Arduino #615: SDK support for BLE isosynchronous audio transmission using nRF52

Protonerd, 2021-01-07:
BL v5.2 supports the so called isosynchronous audio transmission, with which a BLE (as opposed to traditional BL) radio can transmit audio data to a BLE capable external speaker.

Is there any plan to offer an SDK and Arduino package update to support this new version of the standard?

My application I target using the nRF832 shall have audio transmission to an external speaker so I'm interested to know if it can be done with an SDK update or if hardware redesign is necessary (therefore the existing nRF cores will not be able to support the new standard)
https://www.bluetooth.com/blog/10-frequently-asked-questions-on-le-isochronous-channels/

hathach, 2021-01-07:
Short answer is: we have no plan for this. But I will be more than happy to review PR for this.

Protonerd, 2021-01-07:
Thanks hathach for the quick reply, I understand. Just out of curiosity: do you know if this feature needs hardware change or "just" a modification of the SDK from Nordic?

fanoush, 2023-07-31:
> do you know if this feature needs hardware change or "just" a modification of the SDK from Nordic?

came from google search,  some answers here

https://devzone.nordicsemi.com/f/nordic-q-a/88043/nrf52840-isochronous-channel-support
https://devzone.nordicsemi.com/f/nordic-q-a/72791/bluetooth-5-2-support-for-nrf52840/299994#299994
https://devzone.nordicsemi.com/f/nordic-q-a/93689/nrf5340-lc3-decoder-question
https://www.nordicsemi.com/News/2020/01/Nordic-launches-a-Bluetooth-LE-Audio-Evaluation-Platform

TLDR -  SoftDevice won't support it (Zephyr based nRF Connect SDK is needed) but the hardware does support it. For full stereo LC3 encoding the cpu may not be fast enough so nordic targets LE audio for dual core nrf5340, not 52832/40

So it may work with 52832 but will need some changes, even Zephyr RTOS examples may not work out of box with 52832
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/broadcast_audio_source/sample.yaml