adafruit/Adafruit_nRF52_Arduino #616: Ways of disabling LED toggling upon bootloader execution on pins 17 and 19

Protonerd, 2021-01-07:
Dear Adafruit, is there a way to programatically disable the bootloader code which makes the pins 17 and 19 toggle at start-up? In my application I would like to use these two pins for alternate functionalities, which works well, except that during bootloading there are glitches caused by the toggling.

hathach, 2021-01-08:
you could modify the bootlaoder code to do so. Next time, please fill out all the question in the issue template when opening one.

closed 2021-01-08 by hathach

Protonerd, 2021-01-08:
Hi hathach, thanks for your quick feedback. I was not aware I can change the bootloader. Do you have a document explaining what is involved in a bootloader update? In the old 8-bit Arduino ecosystem the bootloaders were provided as a hex file. Is it different for the 32-bit ARM cores?
Where do I find the bootloader code and how can I compile it? After compilation, can I update the bootloader over UART (DFU)?

hathach, 2021-01-08:
It is still the same, you need to compile the bootloader to your need and update it (bootloader is self-upgradable). https://github.com/adafruit/Adafruit_nRF52_Bootloader

There is tutorial in Adafruit learning guide. If you have more issue, please open question in our support forum there is more people there to help you out.