adafruit/Adafruit_nRF52_Arduino #617: How to use function setAddr

deepkap, 2021-01-23:
I want to change MAC address of BLE device runtime. 
I'm doing following steps, but getting error by calling setAddr function. 
Here is my code section:

`Bluefruit.Advertising.stop();
Bluefruit.Advertising.clearData();
Bluefruit.ScanResponse.clearData();
connectionHandle = Bluefruit.connHandle();                
connection = Bluefruit.Connection(connectionHandle);        
delay(1000);
connection->disconnect();
delay(2000);

uint8_t mac[6] = {0};
uint8_t addr_type = Bluefruit.getAddr(mac);
Serial.print("Address Type: ");
Serial.println(addr_type);
for(int i = 0; i < 6; i++)
{
Serial.print(mac[i], HEX);
Serial.print(' ');
}
Serial.println();

ble_gap_addr_t* gap_addr;
gap_addr->addr_type = BLE_GAP_ADDR_TYPE_RANDOM_STATIC;
gap_addr->addr[0] = mac[0];
gap_addr->addr[1] = mac[1];
gap_addr->addr[2] = mac[2];
gap_addr->addr[3] = mac[3];
gap_addr->addr[4] = mac[4];
gap_addr->addr[5] = 0xCE;

if(Bluefruit.setAddr(gap_addr))
{
Serial.println("Addr change Done");
}
else
{
Serial.println("Addr change Error");
}

// Again start advertising`

Bluefruit.setAddr() function return ZERO everytime.

Please guide me how to use setAddr function properly?

hathach, 2021-01-23:
for questions and how-to please post on our support forum

closed 2021-01-23 by hathach

zwells, 2023-07-14:
Is there a way to set the address that works? In terms of defining the ble_gap_addr_t?

  ble_gap_addr_t* gap_addr;
  gap_addr->addr[0] = 0x00;
  gap_addr->addr[1] = 0x00;
  gap_addr->addr[2] = 0x00;
  gap_addr->addr[3] = 0x00;
  gap_addr->addr[4] = 0x00;
  gap_addr->addr[5] = 0x00;
  gap_addr->addr_type = BLE_GAP_ADDR_TYPE_RANDOM_STATIC;
  Bluefruit.setAddr(gap_addr);

^Trying to set it this way before starting to advertise doesn't work.