adafruit/Adafruit_nRF52_Arduino #618: How to use this core with custom boards?

pvonmoradi, 2021-01-24:
Is it possible to use this core with a custom board designed around nRF52832? I'm using PlatformIO.
[arduino-nRF5](https://github.com/sandeepmistry/arduino-nRF5) is a similar project but with older SoftDevice/nrfx version. In that project, we could define a `variant.h` and a `custom_board.json` manifest in PIO.
I'm wondering what are the steps to use Adafruit core for a custom board.

ladyada, 2021-01-24:
hi you can follow this
https://www.instructables.com/Arduino-IDE-Creating-Custom-Boards/
we do not have tutorials on creating custom boards...we are quite busy supporting our own :)

closed 2021-01-24 by ladyada

jpconstantineau, 2021-01-24:
@pvonmoradi The instructions @ladyada linked to are excellent.  If you want a specific example with this core, you can head over to the [Community nRF52 BSP Repo](https://github.com/jpconstantineau/Community_nRF52_Arduino).  I created this core specifically to support the BlueMicro keyboard firmware project.  It's based on 0.21.0 (before the cryptocell changes).