adafruit/Adafruit_nRF52_Arduino #620: Whitelist BLE advertising

jfedor2, 2021-01-28:
Currently when advertising the library accepts connections from any device. It would be nice to be able to use the whitelist feature to restrict the incoming connections to previously bonded peers only. That way it would be possible to replicate the behavior of most consumer Bluetooth devices, where they normally only connect with known peers and you have to specifically request pairing mode by pressing a dedicated button.

hathach, 2022-07-19:
no, but any PRs are welcome

UUSim, 2023-09-17:
This can be easily achieved by putting something like this in the `scan_callback` (or similarly-named) function:

```cpp
void scan_callback(ble_gap_evt_adv_report_t* report) {
    // Check if already bonded
    bond_keys_t peer_bkeys; // We don't need the actual key, just need the struct for calling the function below.
    if (bond_load_keys(BLE_GAP_ROLE_CENTRAL, &report->peer_addr, &peer_bkeys)) {
        Bluefruit.Central.connect(report);
        Bluefruit.Scanner.stop();
        return;
    }
    Bluefruit.Scanner.resume();
```

For pheripheral, adjust the role to `...PERIPHERAL`, and adjust the connect-call.