adafruit/Adafruit_nRF52_Arduino #621: nRF52 GPIO irreversible pin damage when used as a momentary switch sense input

Protonerd, 2021-01-31:
**Problem description:**
when using an nRF52832 GPIO as a momentary switch sense pin, pin can be reproducably damaged.

[(https://devzone.nordicsemi.com/f/nordic-q-a/71117/nrf52832-gpio-irreversible-pin-damage-in-application-when-used-as-a-momentary-switch-sense-input
)](url)

**Use case:** momentary switch connected between a GPIO and GND. Pin configured as input with internal pull up connected (13 kOhms).
Pin is initialized using the function:
`pinMode(mPin, INPUT_PULLUP); //Initialize the pin mode`

**Reprodution details:**
so far I was able to provoke the damage on several of my custom boards and on an Adafruit nRF52 Bluefruit. Problem appeared more often on a pin which was used more often and kept depressed for prolonger periods (~10% of time). Connecting the GPIO to GND after configuring it as input with pull-up should provoke the fail, no records on how long a time is needed though.

**Containment action:** currently I consider this as a potential design bug in the Nordic nRF52 series of BLE chips. Until we get a feedback from the manufacturer the best containment I have is to insert a series resistance from the GPIO to the momentary switch terminal, a 1kohms would do it (considering 3.3V regulated supply and the input compator thresholds, a max of 5kohms could be used) to limit the current. However without knowing the exact physical root cause of the degradation it is hard to tell if this is going to help.

Another line of action I can think of is to look at the nRF52 specific implementation of the _pinMode_ function. The only potential fault mode causing a short to GND internally is if both input and output parts of the GPIO are acrtive and the chip is actively driving a high signal when pulled down to GND by the switch. However as far as I could understand from the data sheet a GPIO can either be used as an input or as an output but not as both at the same time.

This bug report is more of a heads-up, as it can affect a frequent use case and one Adafruit product too.

x

dhalbert, 2021-02-01:
I am following this on DevZone as well.

Our support folks have not seen any reports of pin damage that I know of. We use and recommend the internal pullup as described all the time. However, most customers at this point are using the nRF52840, not nRF52832.

Protonerd, 2021-02-02:
Is there a difference between the 840 and 832? At least the Bluefruit uses the 832 too.

BTW, we also use the internal 13kOhms pull-up.

dhalbert, 2021-02-02:
The '840 has USB support, and much more RAM and flash. I have no idea if it has reworked the physical various peripherals on chip. But we didn't have these reports on the '832 boards either.

I will make some further remarks in DevZone.

hathach, 2021-09-13:
I think we could close this one now since it is mostly hardware issue, and I dont know how we could help with software. If you do please submit an PR for comment or reminder etc…

closed 2021-09-13 by hathach