adafruit/Adafruit_nRF52_Arduino #624: Configuring Systick disables Softdevice

Wookie1, 2021-03-14:
If I set up the Systick feature as below (where I have defined TICK_INTERVAL_MS already), the Bluetooth functionality stops or if it was not already advertising, cannot start.

`SysTick_Config( (F_CPU / 128000) * TICK_INTERVAL_MS);`

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.13 on macOS 11.2.3
 - **BSP**   : 0.21.0
 - **Board** : Feather nRF52832

Working code below, which is the BLEUART example:
`/*********************************************************************
  This is an example for our nRF52 based Bluefruit LE modules

  Pick one up today in the adafruit shop!

  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  MIT license, check LICENSE for more information
  All text above, and the splash screen below must be included in
  any redistribution
*********************************************************************/
#include <bluefruit.h>
#include <Adafruit_LittleFS.h>
#include <InternalFileSystem.h>

#define TICK_INTERVAL_MS 10

// BLE Service
BLEDfu  bledfu;  // OTA DFU service
BLEDis  bledis;  // device information
BLEUart bleuart; // uart over ble
BLEBas  blebas;  // battery

void setup()
{
  Serial.begin(115200);

#if CFG_DEBUG
  // Blocking wait for connection when debug mode is enabled via IDE
  while ( !Serial ) yield();
#endif

  Serial.println("Bluefruit52 BLEUART Example");
  Serial.println("---------------------------\n");

  // Setup the BLE LED to be enabled on CONNECT
  // Note: This is actually the default behaviour, but provided
  // here in case you want to control this LED manually via PIN 19
  Bluefruit.autoConnLed(true);

  // Config the peripheral connection with maximum bandwidth
  // more SRAM required by SoftDevice
  // Note: All config***() function must be called before begin()
  Bluefruit.configPrphBandwidth(BANDWIDTH_MAX);

  Bluefruit.begin();
  Bluefruit.setTxPower(4);    // Check bluefruit.h for supported values
  Bluefruit.setName("Bluefruit52");
  //Bluefruit.setName(getMcuUniqueID()); // useful testing with multiple central connections
  Bluefruit.Periph.setConnectCallback(connect_callback);
  Bluefruit.Periph.setDisconnectCallback(disconnect_callback);

  // To be consistent OTA DFU should be added first if it exists
  bledfu.begin();

  // Configure and Start Device Information Service
  bledis.setManufacturer("Adafruit Industries");
  bledis.setModel("Bluefruit Feather52");
  bledis.begin();

  // Configure and Start BLE Uart Service
  bleuart.begin();

  // Start BLE Battery Service
  blebas.begin();
  blebas.write(100);

  // Set up and start advertising
  startAdv();

  Serial.println("Please use Adafruit's Bluefruit LE app to connect in UART mode");
  Serial.println("Once connected, enter character(s) that you wish to send");
}

void startAdv(void)
{
  // Advertising packet
  Bluefruit.Advertising.addFlags(BLE_GAP_ADV_FLAGS_LE_ONLY_GENERAL_DISC_MODE);
  Bluefruit.Advertising.addTxPower();

  // Include bleuart 128-bit uuid
  Bluefruit.Advertising.addService(bleuart);

  // Secondary Scan Response packet (optional)
  // Since there is no room for 'Name' in Advertising packet
  Bluefruit.ScanResponse.addName();

  /* Start Advertising
     - Enable auto advertising if disconnected
     - Interval:  fast mode = 20 ms, slow mode = 152.5 ms
     - Timeout for fast mode is 30 seconds
     - Start(timeout) with timeout = 0 will advertise forever (until connected)

     For recommended advertising interval
     https://developer.apple.com/library/content/qa/qa1931/_index.html
  */
  Bluefruit.Advertising.restartOnDisconnect(true);
  Bluefruit.Advertising.setInterval(32, 244);    // in unit of 0.625 ms
  Bluefruit.Advertising.setFastTimeout(30);      // number of seconds in fast mode
  Bluefruit.Advertising.start(0);                // 0 = Don't stop advertising after n seconds
}

void loop()
{
  // Forward data from HW Serial to BLEUART
  while (Serial.available())
  {
    // Delay to wait for enough input, since we have a limited transmission buffer
    delay(2);

    uint8_t buf[64];
    int count = Serial.readBytes(buf, sizeof(buf));
    bleuart.write( buf, count );
  }

  // Forward from BLEUART to HW Serial
  while ( bleuart.available() )
  {
    uint8_t ch;
    ch = (uint8_t) bleuart.read();
    Serial.write(ch);
  }
}

// callback invoked when central connects
void connect_callback(uint16_t conn_handle)
{
  // Get the reference to current connection
  BLEConnection* connection = Bluefruit.Connection(conn_handle);

  char central_name[32] = { 0 };
  connection->getPeerName(central_name, sizeof(central_name));

  Serial.print("Connected to ");
  Serial.println(central_name);
}

/**
   Callback invoked when a connection is dropped
   @param conn_handle connection where this event happens
   @param reason is a BLE_HCI_STATUS_CODE which can be found in ble_hci.h
*/
void disconnect_callback(uint16_t conn_handle, uint8_t reason)
{
  (void) conn_handle;
  (void) reason;

  Serial.println();
  Serial.print("Disconnected, reason = 0x"); Serial.println(reason, HEX);
}`

**The working code generates this serial port output below (debug level 2):**
21:34:40.638 -> Bluefruit52 BLEUART Example
21:34:40.638 -> ---------------------------
21:34:40.638 -> 
21:34:40.982 -> [CFG   ] SoftDevice's RAM requires: 0x200034A0
21:34:40.982 -> Please use Adafruit's Bluefruit LE app to connect in UART mode
21:34:40.982 -> Once connected, enter character(s) that you wish to send
21:34:40.982 -> Systick not used...
21:34:40.982 -> ...Setup Complete.
21:34:40.982 -> 
21:34:40.982 -> BSP Library : 0.21.0
21:34:40.982 -> Bootloader  : s132 6.1.1
21:34:40.982 -> Serial No   : 95F47FAB7B97C446
21:34:40.982 -> 
21:34:40.982 -> --------- SoftDevice Config ---------
21:34:41.020 -> Max UUID128     : 10
21:34:41.020 -> ATTR Table Size : 3072
21:34:41.020 -> Service Changed : 1
21:34:41.020 -> Peripheral Connect Setting
21:34:41.020 ->   - Max MTU         : 247
21:34:41.020 ->   - Event Length    : 100
21:34:41.020 ->   - HVN Queue Size  : 3
21:34:41.020 ->   - WrCmd Queue Size: 1
21:34:41.020 -> 
21:34:41.020 -> --------- BLE Settings ---------
21:34:41.020 -> Name            : Bluefruit52
21:34:41.020 -> Max Connections : Peripheral = 1, Central = 0 
21:34:41.020 -> Address         : ED:73:A1:B9:04:EE (Static)
21:34:41.020 -> TX Power        : 4 dBm
21:34:41.053 -> Conn Intervals  : min = 20.00 ms, max = 30.00 ms
21:34:41.053 -> Conn Timeout    : 2000 ms
21:34:41.053 -> Peripheral Paired Devices: 
21:34:41.053 -> 
21:34:41.053 -> 
21:35:10.947 -> [BLE   ] BLE_GAP_EVT_ADV_SET_TERMINATED : Conn Handle = 65535

Adding the Systick configuration as follows (snippet shown here, rest of code is the same):
` // Set up and start advertising
  startAdv();

  Serial.println("Please use Adafruit's Bluefruit LE app to connect in UART mode");
  Serial.println("Once connected, enter character(s) that you wish to send");

  SysTick_Config( (F_CPU / 128000) * TICK_INTERVAL_MS);
  Serial.println("...Setup Complete.");
}`

**Results in this serial port output:**
21:41:27.955 -> Bluefruit52 BLEUART Example
21:41:27.955 -> ---------------------------
21:41:27.955 -> 
21:41:28.272 -> [CFG   ] SoftDevice's RAM requires: 0x200034A0
21:41:28.307 -> Please use Adafruit's Bluefruit LE app to connect in UART mode
21:41:28.307 -> Once connected, enter character(s) that you wish to send

hathach, 2021-03-14:
please put all the code the the block code, making it easier to read your issue.

Wookie1, 2021-03-19:
@hathach done.

hathach, 2021-03-19:
what is your config value, make sure test this out with this example. There is a note there for max period systick can hold roughly 262 ms 
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Hardware/hw_systick/hw_systick.ino#L43

Wookie1, 2021-03-20:
In the above example, it is 10ms. In my application, I use 1ms so if I need to delay() I instead created a function that breaks it into 1ms delays with feeding the watchdog after each one. Either way, though, it is way below 262ms and has been working on BSP 0.19.0. I haven't tried it on any BSP's between 0.19.0 and 0.21.0, so I'm not sure when it broke.

hathach, 2021-03-21:
have you tried the hw example in the link above with your delay and confirm that it works.

hathach, 2021-04-22:
closed due to lack of user response

closed 2021-04-22 by hathach