adafruit/Adafruit_nRF52_Arduino #625: add support for CMSIS arm_math math hardware accelerator on nRF52 chips

LokiMetaSmith, 2021-03-18:
Please consider adding support for arm_math.h and the built in math libraries. 
NRF52832 , NRF52833 , NRF52840 all support built in floating point and matrix math library functions, but currently I have to edit compiler flags deep in the bowles of arduino to get them to work.

https://community.platformio.org/t/project-using-arm-math-h-on-nrf52-results-in-linker-error/7794/8

https://github.com/ARM-software/CMSIS/blob/master/CMSIS/Include/arm_math.h

hathach, 2021-03-19:
would you mind submitting an PR for this, I will be happy to review.

LokiMetaSmith, 2021-03-25:
I reviewed how best to incorporate the CMSIS blob, and it looks like 

adafruit/ArduinoCore-samd implemented CMSIS differently than how adafruit/Adafruit_nRF52_Arduino has it implemented. 
https://github.com/adafruit/ArduinoCore-samd/pull/232/files

My initial method is way too hackish. Also, I couldn't figure out how to generalize it for all boards. I'll push a PR with what I have. 

--------------------------------- platform.txt ---------------------------------
index 3c602a26..ab6675c9 100644
@@ -64,7 +64,7 @@ nordic.path={build.core.path}/nordic
 
 # build.logger_flags and build.sysview_flags are intentionally empty,
 # to allow modification via a user's own boards.local.txt or platform.local.txt files.
-build.flags.nrf= -DSOFTDEVICE_PRESENT -DARDUINO_NRF52_ADAFRUIT -DNRF52_SERIES -DDX_CC_TEE -DLFS_NAME_MAX=64 {compiler.optimization_flag} {build.debug_flags} {build.logger_flags} {build.sysview_flags} "-I{build.core.path}/cmsis/Core/Include" "-I{nordic.path}" "-I{nordic.path}/nrfx" "-I{nordic.path}/nrfx/hal" "-I{nordic.path}/nrfx/mdk" "-I{nordic.path}/nrfx/soc" "-I{nordic.path}/nrfx/drivers/include" "-I{nordic.path}/nrfx/drivers/src" "-I{nordic.path}/softdevice/{build.sd_name}_nrf52_{build.sd_version}_API/include" "-I{nordic.path}/softdevice/{build.sd_name}_nrf52_{build.sd_version}_API/include/nrf52" "-I{rtos.path}/Source/include" "-I{rtos.path}/config" "-I{rtos.path}/portable/GCC/nrf52" "-I{rtos.path}/portable/CMSIS/nrf52" "-I{build.core.path}/sysview/SEGGER" "-I{build.core.path}/sysview/Config" "-I{build.core.path}/TinyUSB" "-I{build.core.path}/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-I{build.core.path}/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src"
+build.flags.nrf= -DSOFTDEVICE_PRESENT -DARDUINO_NRF52_ADAFRUIT -DNRF52_SERIES -DDX_CC_TEE -DLFS_NAME_MAX=64 {compiler.optimization_flag} {build.debug_flags} {build.logger_flags} {build.sysview_flags} "-I{build.core.path}/cmsis/Core/Include" "-I{build.core.path}/cmsis/DSP/Include" "-I{nordic.path}" "-I{nordic.path}/nrfx" "-I{nordic.path}/nrfx/hal" "-I{nordic.path}/nrfx/mdk" "-I{nordic.path}/nrfx/soc" "-I{nordic.path}/nrfx/drivers/include" "-I{nordic.path}/nrfx/drivers/src" "-I{nordic.path}/softdevice/{build.sd_name}_nrf52_{build.sd_version}_API/include" "-I{nordic.path}/softdevice/{build.sd_name}_nrf52_{build.sd_version}_API/include/nrf52" "-I{rtos.path}/Source/include" "-I{rtos.path}/config" "-I{rtos.path}/portable/GCC/nrf52" "-I{rtos.path}/portable/CMSIS/nrf52" "-I{build.core.path}/sysview/SEGGER" "-I{build.core.path}/sysview/Config" "-I{build.core.path}/TinyUSB" "-I{build.core.path}/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-I{build.core.path}/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src"
 
 # usb flags
 build.flags.usb= -DUSBCON -DUSE_TINYUSB -DUSB_VID={build.vid} -DUSB_PID={build.pid} '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT={build.usb_product}'
@@ -75,11 +75,12 @@ compiler.c.elf.extra_flags=
 compiler.cpp.extra_flags=
 compiler.S.extra_flags=
 compiler.ar.extra_flags=
-compiler.libraries.ldflags=
+compiler.libraries.ldflags={build.core.path}/cmsis/DSP/Lib/GCC/libarm_cortexM4lf_math.a
 compiler.elf2bin.extra_flags=
 compiler.elf2hex.extra_flags=
 
 
+
 # Compile patterns
 # ----------------

PR #658 (2021-06-21): Port TensorFlow Lite Motion Kit project
This PR port Tiny Motion Trainer https://experiments.withgoogle.com/tiny-motion-trainer. Changes is quite a lots
- Update CMSIS from 5.4.0 to `5.7.0` to build with latest TensorFlowLite v2.4.0
- Enable dsp libmath to build with latest TensorFlowLite v2.4.0 (close #626)
- Implent BLEUuid to allow initialize Characteristic and Service following ArduinoBLE style
- Implement Server/Characteristic constructor
```C
    BLECharacteristic(BLEUuid bleuuid, uint8_t properties);
    BLECharacteristic(BLEUuid bleuuid, uint8_t properties, int max_len, bool fixed_len  [... cut]
closed 2021-06-21 by hathach