adafruit/Adafruit_nRF52_Arduino #628: peripheral pairing_pin Example doesn't compile

ericlangel, 2021-04-07:
**Describe the bug**
peripheral pairing_pin Example doesn't compile
there are many headers not found 

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.13 Windows 10
 - **BSP**   : 0.21.0 (Master)
 - **Board** : Feather nRF52832
 - **Sketch**: Bluefruit52Lib/examples/Peripheral/pairing_pin

**To Reproduce**
compile sketch with actual master branch

**Serial Log**

Open the sketch:
Ungültige Bibliothek C:\Users\user\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\libraries\Adafruit_nRFCrypto in keine Header-Dateien (.h) in C:\Users\user\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\libraries\Adafruit_nRFCrypto gefunden gefunden
Ungültige Bibliothek C:\Users\user\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\libraries\Adafruit_nRFCrypto in keine Header-Dateien (.h) in C:\Users\user\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\libraries\Adafruit_nRFCrypto gefunden gefunden
Ungültige Bibliothek C:\Users\user\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\libraries\Adafruit_nRFCrypto in keine Header-Dateien (.h) in C:\Users\user\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\libraries\Adafruit_nRFCrypto gefunden gefunden

Compile the Sketch:
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\eric\AppData\Local\Arduino15\packages -hardware C:\Users\eric\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\eric\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\eric\Documents\Arduino\libraries -fqbn=Adafruit:Adafruit_nRF52_Arduino-master:feather52832:softdevice=s132v6,debug=l0 -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\eric\AppData\Local\Temp\arduino_build_966391 -warnings=default -build-cache C:\Users\eric\AppData\Local\Temp\arduino_cache_573690 -prefs=build.warn_data_percentage=75 -verbose C:\Users\eric\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\libraries\Bluefruit52Lib\examples\Peripheral\pairing_pin\pairing_pin.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\eric\AppData\Local\Arduino15\packages -hardware C:\Users\eric\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\eric\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\eric\Documents\Arduino\libraries -fqbn=Adafruit:Adafruit_nRF52_Arduino-master:feather52832:softdevice=s132v6,debug=l0 -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\eric\AppData\Local\Temp\arduino_build_966391 -warnings=default -build-cache C:\Users\eric\AppData\Local\Temp\arduino_cache_573690 -prefs=build.warn_data_percentage=75 -verbose C:\Users\eric\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\libraries\Bluefruit52Lib\examples\Peripheral\pairing_pin\pairing_pin.ino
Using board 'feather52832' from platform in folder: C:\Users\eric\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master
Using core 'nRF5' from platform in folder: C:\Users\eric\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master
Detecting libraries used...
"C:\\Users\\eric\\AppData\\Local\\Arduino15\\packages\\RFduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=64000000 -DARDUINO=10813 -DARDUINO_NRF52832_FEATHER -DARDUINO_ARCH_ADAFRUIT_NRF52_ARDUINO-MASTER "-DARDUINO_BSP_VERSION=\"0.21.0\"" -DNRF52832_XXAA -DNRF52 -DSOFTDEVICE_PRESENT -DARDUINO_NRF52_ADAFRUIT -DNRF52_SERIES -DDX_CC_TEE -DLFS_NAME_MAX=64 -Ofast -DCFG_DEBUG=0 -DCFG_LOGGER=1 -DCFG_SYSVIEW=0 "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/cmsis/Core/Include" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/nordic" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/nordic/nrfx" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/nordic/nrfx/hal" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/nordic/nrfx/mdk" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/nordic/nrfx/soc" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/nordic/nrfx/drivers/include" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/nordic/nrfx/drivers/src" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/nordic/softdevice/s132_nrf52_6.1.1_API/include" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/nordic/softdevice/s132_nrf52_6.1.1_API/include/nrf52" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/freertos/Source/include" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/freertos/config" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/freertos/portable/GCC/nrf52" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/freertos/portable/CMSIS/nrf52" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/sysview/SEGGER" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/sysview/Config" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/TinyUSB" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\cores\\nRF5" "-IC:\\Users\\eric\\Documents\\Arduino\\hardware\\Adafruit\\Adafruit_nRF52_Arduino-master\\variants\\feather_nrf52832" "C:\\Users\\eric\\AppData\\Local\\Temp\\arduino_build_966391\\sketch\\pairing_pin.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for common/tusb_compiler.h: []
ResolveLibrary(common/tusb_compiler.h)
  -> candidates: []
In file included from C:\Users\eric\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\cores\nRF5/common_inc.h:43:0,
                 from C:\Users\eric\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\cores\nRF5/HardwarePWM.h:39,
                 from C:\Users\eric\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\cores\nRF5/Arduino.h:58,
                 from C:\Users\eric\AppData\Local\Temp\arduino_build_966391\sketch\pairing_pin.ino.cpp:1:
C:\Users\eric\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\cores\nRF5/compiler_macro.h:42:34: fatal error: common/tusb_compiler.h: No such file or directory
 #include "common/tusb_compiler.h"
                                  ^
compilation terminated.
exit status 1
Fehler beim Kompilieren für das Board Adafruit Feather nRF52832.
Ungültige Bibliothek C:\Users\eric\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\libraries\Adafruit_nRFCrypto in keine Header-Dateien (.h) in C:\Users\eric\Documents\Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino-master\libraries\Adafruit_nRFCrypto gefunden gefunden

ericlangel, 2021-04-07:
same for bleuart example

common/tusb_compiler.h is missing?

is there a working example for pairing_pin?

hathach, 2021-04-07:
firstly please clean up your 1st post and have compile log in its own txt file.

jpconstantineau, 2021-04-07:
Quick comment: the nrf52832 does not have a cryptocell nor USB.  Why then is this trying to compile wit Adafruit_nRFCrypto and Tinyusb?  This is probably why it's not compiling...

ericlangel, 2021-04-08:
@hathach sorry, i did clean it

@jpconstantineau same for Feather NRF52840 Express

hathach, 2021-04-22:
@ericlangel you need to clone the TinyUSB core submodule as well, download zip file won't do the trick.

```
Clone this repo & its submodules: git clone --recurse-submodules https://github.com/adafruit/Adafruit_nRF52_Arduino.git
```

closed 2021-04-22 by hathach