adafruit/Adafruit_nRF52_Arduino #635: cores/nRF5/common_func.h uses undefined size_t (missing stddef.h include in common_inc.h)

seaeagle1, 2021-05-08:
**Describe the bug**
While trying to compile a sketch using the nRFCrypto library, I got an error about undefined size_t types in cores/nRF5/common_func.h. nRFCrypto uses the cores/nRF5/common_inc.h header which includes stdint.h and stdbool.h before common_func.h but lacks the stddef.h that defines size_t. Adding this include here solves the problem.

**Set up (mandatory)**
 - **PC & IDE** : Arduino 2.0.0beta5 on Windows 10
 - **BSP**   : 0.21.0
 - **Board** : ItsyBitsy nRF52840 Express
 - **Sketch**: nRFCrypto

hathach, 2021-05-09:
Can you tell which specific sketch you are compiling

seaeagle1, 2021-05-10:
An example is the Bluefruit Clear Bonds example.

hathach, 2021-06-03:
please try to update to latest BSP, I couldn't reproduce this

hathach, 2021-06-30:
Closed since not reproducible

closed 2021-06-30 by hathach