adafruit/Adafruit_nRF52_Arduino #638: Adafruit ItsiBitsy nRF52840 Express hangs completely when using Fabrik2D library

xchg-dot-ca, 2021-05-14:
**Describe the bug**
When using Fabrik2D library and trying to solve inverse kinematic scenario board hangs and requires hold reset procedure to reflsah.
I assume board enters a infinite loop. I believe there is some kind of imperfection in double/float math library of the board.
This issue is NOT happening on: Arduino UNO, Arduino Leonardo, Teensy 3.1, Arduino Due

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.13 on Windows 7
 - **BSP**   : 0.2.11

 - **Board** : Adafruit ItsiBitsy nRF52840 Express
 - **Sketch**: Test sketch attached

**To Reproduce**
Please install attached library or install it via the Library manager of the IDE
Compile and Upload attached sketch
[example_3DOFToolAngle.zip](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/6479518/example_3DOFToolAngle.zip)
[Fabrik2D.zip](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/6479523/Fabrik2D.zip)

**Screenshots**
n/a

**Serial Log**
n/a

xchg-dot-ca, 2021-07-19:
Disregard, issue was specifically with library not having return statement:
https://github.com/henriksod/Fabrik2DArduino/pull/20

Still interesting to see different behaviour for Adafruit library that it getting lost if not return statement provided in the code
In same time Arduino products work fine

hathach, 2021-07-19:
I didn't look at the code, for this core, if you have 

```C
while(1) {}
```

it will entirely occupy the CPU and does not give any other task such as tinyusb to handle any printf, or touch1200. Making it is look like hanged. It is probably the case here. What you should do is adding either yield() or delay(1) in your blocking.

```C
while(1) yield(); // or delay(1)
```

hathach, 2021-09-13:
Could be closed now

closed 2021-09-13 by hathach