adafruit/Adafruit_nRF52_Arduino #643: error: 'atomic_uint32_t' in namespace 'std' does not name a type

NilsMinor, 2021-05-20:
**Describe the bug**
After updating to v0.22.0 i get the following error message when trying to compile my project

`
In file included from /Users/nilsminor/Library/Arduino15/packages/adafruit/hardware/nrf52/0.22.0/cores/nRF5/Arduino.h:58:0,
                 from /Users/nilsminor/Documents/code/nils/huli/huli-one/firmware/huli-os/huli-os.ino:1:
/Users/nilsminor/Library/Arduino15/packages/adafruit/hardware/nrf52/0.22.0/cores/nRF5/HardwarePWM.h:54:5: error: 'atomic_uint32_t' in namespace 'std' does not name a type
     std::atomic_uint32_t _owner_token;
     ^
/Users/nilsminor/Library/Arduino15/packages/adafruit/hardware/nrf52/0.22.0/cores/nRF5/HardwarePWM.h: In member function 'bool HardwarePWM::isOwner(uint32_t) const':
/Users/nilsminor/Library/Arduino15/packages/adafruit/hardware/nrf52/0.22.0/cores/nRF5/HardwarePWM.h:81:20: error: 'const class HardwarePWM' has no member named '_owner_token'
       return this->_owner_token == token;
                    ^
exit status 1
`

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.13 macOS 11.2.3 
 - **BSP**   : 0.22.0
 - **Board** : Feather nRF52832 Express

My compiler args are called from viscose like this

`
       "name": "Arduino",
            "compilerPath": "/Users/nilsminor/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++",
            "compilerArgs": [
                "-mcpu=cortex-m4",
                "-mthumb",
                "-Wall",
                "-Wextra",
                "-Wno-unused-parameter",
                "-Wno-missing-field-initializers",
                "-Wno-pointer-arith",
                "-mfloat-abi=hard",
                "-mfpu=fpv4-sp-d16",
                "-u",
                "-std=gnu++11",
                "-ffunction-sections",
                "-fdata-sections",
                "-fno-threadsafe-statics",
                "-nostdlib",
                "--param",
                "-fno-rtti",
                "-fno-exceptions"
            ],
            "intelliSenseMode": "gcc-x64",
`
Thanks for your help,
Nils

hathach, 2021-05-21:
Which sketch you are compiling and how to reproduce the issue ?

NilsMinor, 2021-05-22:
This compile error comes on every Bluefruit sketch, for example if I run the bleuart.ino example. I think this error message comes due to my compiler settings but I am not sure how to fix this. Any Idea what I need to change? I just updated from V0.21.0 to V0.22.0 and the error message occurred.

hathach, 2021-05-22:
closed since I cannot reproduce this issue with Arduino IDE, for changes in 0.22, you could checkout the commit log along with the release note.

closed 2021-05-22 by hathach