adafruit/Adafruit_nRF52_Arduino #646: Update compile flag to c++14

jpconstantineau, 2021-05-25:
**Is your feature request related to a problem? Please describe.**
I want to add a c++ library that depends on C++14 features.  Upgrading from c++11 to c++14 should take care of that.

**Describe the solution you'd like**
Update the platform.txt file to update the gcc command like flags.
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/platform.txt#L43

**Describe alternatives you've considered**
I have tested it here locally on a fork and haven't seen any issues (c++14 should be backward compatible with c++11).

**Additional context**
If you are OK with going ahead, I can make the change here and create a PR.  I just want to know if you would approve the update as it could impact other things I may not be aware of.  The gcc arm compiler has been supporting this flag for quite a while.

jpconstantineau, 2021-05-25:
This would be done similarly to this PR: https://github.com/SpenceKonde/ATTinyCore/pull/558/commits/d5c5cf6927716c4a39b821ba09767df16da6da5b

hathach, 2021-05-26:
are you also using gcc 9 as the bsp. If you feel free to submit an PR, however, it may not be merged immediately. There is a quite pending number of issues I would like to resolve first and changing compiler dialect could complicate the issue.

jpconstantineau, 2021-06-04:
That probably wise.  Also, I would probably recommend taking the smaller step of going to 14 first as many people indicate that mostly a "bug fix" of c++11.  c++17 brought more things...  There are even talks about some gotchas of moving to c++17.  https://youtu.be/mAZyaAo3M70

jpconstantineau, 2021-06-05:
@hathach I just started testing 0.22.1 with my project and quite a few errors are coming out.  Some from code that hasn't changed in years.  Other places is new with the 0.22.x releases.