adafruit/Adafruit_nRF52_Arduino #652: Cannot link any BLE project for nRF52840-based boards

valeros, 2021-05-28:
Hi there, 

If I try to compile any BLE-based examples, it fails on the linking stage with the following error:

**Serial Log**
```
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Temp\arduino_build_591592\libraries\Bluefruit52Lib\BLESecurity.cpp.o: in function `BLESecurity::BLESecurity()':
C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:79: undefined reference to `nRFCrypto_ECC_PrivateKey::nRFCrypto_ECC_PrivateKey()'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Temp\arduino_build_591592\libraries\Bluefruit52Lib\BLESecurity.cpp.o: in function `BLESecurity::begin()':
C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:91: undefined reference to `Adafruit_nRFCrypto::begin()'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:94: undefined reference to `nRFCrypto_ECC_PrivateKey::begin(CRYS_ECPKI_DomainID_t)'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:97: undefined reference to `nRFCrypto_ECC_PublicKey::nRFCrypto_ECC_PublicKey()'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:98: undefined reference to `nRFCrypto_ECC_PublicKey::begin(CRYS_ECPKI_DomainID_t)'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:101: undefined reference to `nRFCrypto_ECC::genKeyPair(nRFCrypto_ECC_PrivateKey&, nRFCrypto_ECC_PublicKey&)'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:104: undefined reference to `nRFCrypto_ECC_PublicKey::toRaw(unsigned char*, unsigned long)'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:105: undefined reference to `nRFCrypto_ECC_PublicKey::end()'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:114: undefined reference to `nRFCrypto'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Temp\arduino_build_591592\libraries\Bluefruit52Lib\BLESecurity.cpp.o: in function `BLESecurity::_eventHandler(ble_evt_t*)':
C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:330: undefined reference to `nRFCrypto_ECC_PublicKey::nRFCrypto_ECC_PublicKey()'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:331: undefined reference to `nRFCrypto_ECC_PublicKey::begin(CRYS_ECPKI_DomainID_t)'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:332: undefined reference to `nRFCrypto_ECC_PublicKey::fromRaw(unsigned char*, unsigned long)'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:336: undefined reference to `nRFCrypto_ECC::SVDP_DH(nRFCrypto_ECC_PrivateKey&, nRFCrypto_ECC_PublicKey&, unsigned char*, unsigned long)'
c:/users/test/appdata/local/arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\test\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.22.1\libraries\Bluefruit52Lib\src/BLESecurity.cpp:338: undefined reference to `nRFCrypto_ECC_PublicKey::end()'

collect2.exe: error: ld returned 1 exit status
```

**Setup**
 - **PC & IDE** : Arduino 1.8.12 on Windows 10
 - **BSP**   : 0.22.1
 - **Board** : Feather nRF52840 Express
 - **Sketch**: Bluefruit52Lib/examples/Peripheral/bleuart.

Am I missing something?

hathach, 2021-05-28:
you miss the part to also clone the submodules of the repos

closed 2021-05-28 by hathach

valeros, 2021-05-28:
Could you please elaborate on what submodules should I clone? I installed the platform using the Arduino IDE so I guess all submodules should be installed by default? You probably mean the `Adafruit_nRFCrypto` library, but I can see the files:
[image]

hathach, 2021-05-28:
sorry, I thought you get it via git clone. which is easy to miss. Your issue requires to update arduino IDE to at least 1.8.13, they change how the IDE link with libraries that come with pre-compiled lib file

hathach, 2021-05-29:
@valeros let me know if updating IDE works for you. Else I will take more closer look at the issue

valeros, 2021-05-31:
@hathach Thanks, updating to the latest IDE resolved the issue.

hathach, 2021-05-31:
I am glad that works out for you