adafruit/Adafruit_nRF52_Arduino #653: Undefined reference to `Serial'

rei-vilo, 2021-06-01:
**Describe the bug**
Adding Serial to the *blinky* example results in error 

>  /tmp/arduino_modified_sketch_618918/Blink.ino:41: undefined reference to `Serial'

**Set up (mandatory)**
 - **PC & IDE** : Arduino 1.8.15 on Debian 10
 - **BSP**   : 0.22.1
 - **Board** : Feather nRF52840 Express
 - **Sketch**: blink. If it is a custom sketch, please provide a minimal sketch within codeblock

**To Reproduce**
Steps to reproduce the behavior:
1. Load example blink
2. Add Serial funcitons
2. Click on 'Verify'
3. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Code**

```
// the setup function runs once when you press reset or power the board
void setup() {
  Serial.begin(115200);

  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {

  Serial.println(millis());

  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

**Serial Log**

> /home/USER/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_603469/sketch/Blink.ino.cpp.o: in function `setup':
> /tmp/arduino_modified_sketch_618918/Blink.ino:27: undefined reference to `Adafruit_USBD_CDC::begin(unsigned long)'
> /home/USER/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_modified_sketch_618918/Blink.ino:30: undefined reference to `Serial'
> /home/USER/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_603469/sketch/Blink.ino.cpp.o: in function `loop':
> /tmp/arduino_modified_sketch_618918/Blink.ino:41: undefined reference to `Serial'
> collect2: error: ld returned 1 exit status
> exit status 1
> Error compiling for board Adafruit Feather nRF52840 Express.

rei-vilo, 2021-06-01:
Release 0.21.0 works fine.

hathach, 2021-06-02:
you need to explicitly add tinyusb header to the sketch since tinyusb is moved from core to libraries entirely.

```C
#include "Adafruit_TinyUSB.h"
```

This is normally not needed when using most of other libraries such as wire, spi, bluefruit etc .. since they include the tinyusb header by default.

rei-vilo, 2021-06-02:
Thank you! It works now.

Would it be possible to migrate the `#include "Adafruit_TinyUSB.h"`
into **Arduino.h** to prevent this false error?

hathach, 2021-06-02:
I would love to, but it is kind of complicated, here is my comment on rp2040 port See https://github.com/earlephilhower/arduino-pico/issues/167#issuecomment-848622174 . The include got its way into most of libraries used by nrf52

```
It won't work, somehow IDE will refuse to link to the link if you library header inside the Arduino.h. It will treat the library as part of the core somehow. Have tried it, maybe you have a better luck.

PS: to be precise, if the path to Tinyusb src is part of the includes, Arduino won’t try to link with the lib (consider it as part of core). Maybe there is a walkaround for this. Will give a try later on. For now, please treat it as SPI or Wire. You need to include it in sketch for using
```

closed 2021-06-02 by hathach

rei-vilo, 2021-06-02:
Thank you for the explanations.