adafruit/Adafruit_nRF52_Arduino #660: Only single service connection between two boards?

antonbrams, 2021-06-11:
Hi, i've following configuration:

"ItsyBItsy on the Left" {
- is defined as a PERIPHERAL to communicate with the computer
-- running homemade service 0x0003
--- running its characteristics at 0x0004 (notify)
-- running another service 0x0005 
--- characteristics at 0x0006 (notify, write)
--- characteristics at 0x0007 (notify, write)
- aswell defined as a CENTRAL to connect to another "ItsyBitsy on the Right"
-- want to connect to the service 0x0003
--- connect to characteristics at 0x004 (notify)
-- connect to the service 0x0005 
--- characteristics at 0x0006 (notify, write)
--- characteristics at 0x0007 (notify, write)
}

"Itsy Bitsy on the Right" {
- is defined as a peripheral to communicate with the "ItsyBitsy on the Left"
-- running homemade service 0x0003
--- running its characteristics at 0x0004 (notify)
-- running another service 0x0005 
--- characteristics at 0x0006 (notify, write)
--- characteristics at 0x0007 (notify, write)
}

I want to send a file to the left one, and from the left transmit to the right itsybity.
If i enable on both boards a single service, everything works. Left half refuses to connect to more than one service of the right board.

Is it a limitation of the system, or is it a bug?

- working on the mac 11.4, arduino 1.8.15
- boards 2x itsybitsy nrf52840

hathach, 2021-06-12:
have you tried the dual role example which does something similar.  https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/DualRoles/dual_bleuart/dual_bleuart.ino

antonbrams, 2021-06-26:
Thank you for your answer. I still can't get it to work. On the way of testing i've found out that my Itsy Bitsy on the left should have same addresses in peripheral and central, otherwise it will interfere with itself and receive sent data )... anyway i've implemented a single service and put all the characteristics in it... not an intended ble design but heck it works and i move on with my project... if somebody had the same issue, then say it aloud ) thank you for your time

hathach, 2021-06-27:
There is more people in support forum to help you there. And you should start with stock example then modify it. BLE is complicated

closed 2021-06-27 by hathach

jpconstantineau, 2021-06-27:
@antonbrams For my BlueMicro_BLE firmware, I have implemented a custom service for communication between halves of a split keyboard.  Unfortunately, not as two services.  Have you tried connecting to standard services and not custom ones?  For example one custom and the battery service? Or one custom and one Blueart?

antonbrams, 2021-06-28:
> @antonbrams For my BlueMicro_BLE firmware, I have implemented a custom service for communication between halves of a split keyboard. Unfortunately, not as two services. Have you tried connecting to [...]

ah no, had no time to experiment so much. Even if it would work, the custom services that i need so much will still be unconnected )

hathach, 2021-06-28:
enable debug level to 2 (via menu) could help you to understand more what activities behind the scene, and error if any