adafruit/Adafruit_nRF52_Arduino #661: nRF52840 hang in vTaskSwitchContext()

willdavrow, 2021-06-15:
I have several nRF52840 boards that I am using as a ble central to connect to two devices. The central seems to hang completely randomly, some days this will not occur at all and at other times it can happen several times within a 5 minute period. I have attached a j link and can see that the HardFault is coming from vTaskSwitchContext().

I program the devices through platformio and they work perfectly except for when the hang occurs.

I have already attempted to increase the stack depth of the rtos and timers by modifying line 58 of FreeRTOSConfig.h as follows:
`#define configTOTAL_HEAP_SIZE                                    ( 8192 ) /* not used since we use malloc */`
as well as line 94 of the same file:
`#define configTIMER_TASK_STACK_DEPTH                             ( 512 )`
and line 84 of rtos.h to:
`#define SCHEDULER_STACK_SIZE_DFLT   (512*4)`
but these changes have not helped. 

I would be very grateful if anyone has any ideas on how to resolve this issue.

hathach, 2021-06-16:
can you clarify this as question or bug issue. For question please post it in forum support as mentioned in the template. If you are sure it is a bug, please use the bug template and provided all information in the required field.

hathach, 2021-06-28:
closed due to lack of response from user

closed 2021-06-28 by hathach