adafruit/Adafruit_nRF52_Arduino #665: BLE HID blocks loop while sending keyPress

antonbrams, 2021-06-26:
Hi, I'm working on ItsyBitsy nrf and trying to implement full featured keyboard. If i hit multiple keys at once, keyPress execution takes a while and blocks the loop so the whole system freezes until all keys are sent.

```
hid.keyPress(key);
hid.keyRelease();
```

hathach, 2021-06-27:
For bus reports please update your post to provide all information required by bug template. For questions, please post it to our forum support

jpconstantineau, 2021-06-27:
@antonbrams Does the same occur if you use the HID report directly instead of the keypress/keyrelease functions?  My BlueMicro keyboard firmware uses the HID report functions and I don't seem to be suffering from this.  Have a look [here](https://github.com/jpconstantineau/BlueMicro_BLE) for more details.

antonbrams, 2021-06-28:
hey, thank you for your attention. I've googled a lot, and found out that the higher bandwidth
`
Bluefruit.configPrphBandwidth(BANDWIDTH_MAX);
Bluefruit.Periph.setConnIntervalMS(15, 30);
`
and disabling persisting scanning
`
Bluefruit.Scanner.start(100);
`
does the job. I've set this settings and the loop function seems to run smoothly now. I don't quite understand why the scanner has to be limited to 100 seconds or what ever, if it shuts down as long I'm connected to other peripherals... On the other hand the bandwidth_max makes the file upload over custom characteristics from noble (mac) to the arduino unstable... Anyway the lack of adafruits documentation makes me sit until late night and loose hairs :/

antonbrams, 2021-06-28:
> @antonbrams Does the same occur if you use the HID report directly instead of the keypress/keyrelease functions? My BlueMicro keyboard firmware uses the HID report functions and I don't seem to be [...]

yeah, i've searched the whole internet for various reasons and have seen your code :) thanks for advice, i've used same settings (i think so by quick scan of your code). For me it doesn't matter which function am i using...  i'm developing innovative all-in-one keyboard without physical keys, so my sensors has to run super super fast to interpret, what the user is doing with his fingers in real time. And any delay in the loop ruins the whole experience :)

eugeneanikin, 2021-06-30:
I am seeing something similar in my project. The delay seems to vary depending on connection - sometimes it is negligible, other times extremely sluggish even to a point of freezing my Linux box. I've been blaming crappy BT driver from Intel, but maybe it is on the nrf side. Good catch about the scanner timeout, may be it will help us find a culprit. I'll poke around...

hathach, 2021-07-01:
I could look at this issue, but first, OP need to update the post or others need to file an new issue to provide more information as well of way to reproduce the issue.

hathach, 2021-08-17:
closed since issue lack reproducible instruction, please open an new bug using template form if it is still an issue.

closed 2021-08-17 by hathach