adafruit/Adafruit_nRF52_Arduino #670: Bond keys are saved to incorrect mac

davejwatson24, 2021-07-03:
### Operating System

Linux

### IDE version

1.8.15

### Board

 nrf52840 DK

### BSP version

0.24.0

### Sketch

HID keyboard example

### What happened ?

Trying to connect HID keyboard example to linux: works fine for first connect.  If I reset the device, it conntinually connects and disconnects, with btmon error message invalid PIN.

Very suspiciously, the debug log says the bond keys are saved to mac all zeros:
```

[BOND  ] Saved keys for "dave/adafruit/bond_prph/000000000000" to file /adafruit/bond_prph/0000)
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 0
[GATTS ] attr's cccd = 0x0001
[BOND  ] Saved CCCD to file /adafruit/bond_prph/000000000000 ( offset = 147, len = 44 bytes )
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 0                                
[GATTS ] attr's cccd = 0x0001                                                 
[BOND  ] Saved CCCD to file /adafruit/bond_prph/000000000000 ( offset = 147, len = 44 bytes )
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 0                                
[GATTS ] attr's cccd = 0x0001                                                 
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BOND  ] Saved CCCD to file /adafruit/bond_prph/000000000000 ( offset = 147, len = 44 bytes )
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 0
[GATTS ] attr's cccd = 0x0001
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BOND  ] Saved CCCD to file /adafruit/bond_prph/000000000000 ( offset = 147, len = 44 bytes )
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BLE   ] BLE_GAP_EVT_DISCONNECTED : Conn Handle = 0
[GAP   ] Disconnect Reason: REMOTE_USER_TERMINATED_CONNECTION

```
If I set a security callback and set the _bond_keys manually, it works correctly:

 ```
 Bluefruit.Security.setSecuredCallback(connection_secured_callback);
...
void connection_secured_callback(uint16_t conn_handle)
{
  BLEConnection* conn = Bluefruit.Connection(conn_handle);

  ble_gap_addr_t addr = conn->getPeerAddr();
  Bluefruit.Security._bond_keys.peer_id.id_addr_info = addr;
...

```

I don't actually see where _bond_keys is otherwise set.  It seems to work for some centrals (iphone) but not others (linux).  Not sure what the real solution is.

### How to reproduce ?

Not sure which centrals cause zeroed mac, but just the hid_keyboard sketch breaks with no changes.

### Debug Log

[BOND  ] Saved keys for "dave/adafruit/bond_prph/000000000000" to file /adafruit/bond_prph/0000)
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 0
[GATTS ] attr's cccd = 0x0001
[BOND  ] Saved CCCD to file /adafruit/bond_prph/000000000000 ( offset = 147, len = 44 bytes )
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 0                                
[GATTS ] attr's cccd = 0x0001                                                 
[BOND  ] Saved CCCD to file /adafruit/bond_prph/000000000000 ( offset = 147, len = 44 bytes )
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 0                                
[GATTS ] attr's cccd = 0x0001                                                 
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BOND  ] Saved CCCD to file /adafruit/bond_prph/000000000000 ( offset = 147, len = 44 bytes )
[BLE   ] BLE_GATTS_EVT_WRITE : Conn Handle = 0
[GATTS ] attr's cccd = 0x0001
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BOND  ] Saved CCCD to file /adafruit/bond_prph/000000000000 ( offset = 147, len = 44 bytes )
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[SOC   ] NRF_EVT_FLASH_OPERATION_SUCCESS
[BLE   ] BLE_GAP_EVT_DISCONNECTED : Conn Handle = 0
[GAP   ] Disconnect Reason: REMOTE_USER_TERMINATED_CONNECTION

hathach, 2021-07-04:
hmm, seems like nrf52 couldn't get the peer mac address, which Linux distro and version you are testing with. Also please attached your full log as txt file for analysis.

davejwatson24, 2021-07-04:
```
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 21.04
Release:	21.04
Codename:	hirsute

```

nrf debug log, btmon log, bluetoothctl sessions all attached.  Keys were erased before this test with bond_clear_all.  This was unmodified hid_keyboard example.
[bluetoothctl.txt](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/6760322/bluetoothctl.txt)
[btmon_log.txt](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/6760324/btmon_log.txt)
[log.txt](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/6760325/log.txt)

hathach, 2021-07-05:
I am able to reproduce the issue, the root cause is that Linux machine doesn't provide its IRK key for resolving ID Address, which I have assumed all bluetooth device would do after securing the connection due to mostly testing with Android/iOS device. 

The reason is Linux PC use Public MAC which is a static (allow other device to track its position), however, since it is most likely desktop, it doesn't seem to care about. I will make an PR update to use correct MAC depending on the type/exchanged info. 

The issue with using the mac of previous connection is just a side-effect, since that filed for resolvable address is not updated when connecting to Linux PC.

hathach, 2021-07-05:
#671  should fix this issue, please try it out and let me know if  that works for you.

PR #671 (2021-07-06): fix bonding with public/static address
Fix issue when bonding with peer device using public address where the IRK+address are not exchanged when pairing (since device think it is not necessary). Should fix #670
closed 2021-07-06 by hathach