adafruit/Adafruit_nRF52_Arduino #672: Issues compiling Blink with latest TinyUSB

eugeneanikin, 2021-07-06:
### Operating System

MacOS

### IDE version

Arduino 1.8.15

### Board

Clue, nrf52840 express, etc

### BSP version

github latest

### Sketch

Examples/Basics/Blink

### What happened ?

Checked out latest Adafruit_nRF52_Arduino with submodules, tried to upload to my board. The testcase seemed to compile ok and uploaded to the board via USB. The sketch appeared to work ok, but /dev/cu.usbmodem143201 disappeared. Hmmmmm!

Added this line to setup():

  Serial.begin(115200);

Tried compiling and got:

/System/Volumes/Data/qnap/Public/Eugene/Arduino/BlinkRGB/Blink.ino:28: undefined reference to `Adafruit_USBD_CDC::begin(unsigned long)'

All right, I add following include:

#include <Adafruit_TinyUSB.h>

I get:

/System/Volumes/Data/qnap/Public/Eugene/Arduino/hardware/Adafruit/Adafruit_nRF52_Arduino/libraries/Adafruit_TinyUSB_Arduino/src/tusb_config.h:48:4: error: #error TinyUSB Arduino Library does not support your core yet
   48 |   #error TinyUSB Arduino Library does not support your core yet

Fine, I add following to tusb_config.h:
#elif defined(ARDUINO_ARCH_ADAFRUIT_NRF52_ARDUINO)
  #include "arduino/ports/nrf/tusb_config_nrf.h"

and it finally starts to compile!
OKey, I've got a workaround, but I'm not sure what's the proper fix.

 

### How to reproduce ?

see above

hathach, 2021-07-06:
Hmm, my guess is arduino IDE seem to use the folder name in your case Adafruit_nRF52_Arduino (since installed via github) for ARCH name, normally installed it via board manage will have the ARCH_NRF52. I will use another defined for this in the next  release

hathach, 2021-07-06:
I have put up PR that should fix this issue, would you mind trying it out, the only update is tinyusb submodule https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/673

eugeneanikin, 2021-07-07:
Looks good for TinyUSB error! 
I'm still seeing the undefined reference from the linker if I have Serial.begin() but comment out the #include, is this intentional?

hathach, 2021-07-07:
> Looks good for TinyUSB error!
> I'm still seeing the undefined reference from the linker if I have Serial.begin() but comment out the #include, is this intentional?

yes, it is intentional, starting from for BSP 0.22.0, tinyusb is moved entirely from core to library, and including the USB is required even if you are only using Serial. Most libraries within the core would include that by default to ease the user, 

but for most basic example like blink, you need to do it in your sketch

PR #673 (2021-07-07): bump up tinyusb to fix build issue with 672
fix #672
closed 2021-07-07 by hathach