adafruit/Adafruit_nRF52_Arduino #674: Compile error including delay.h

pyro9, 2021-07-12:
In a library, I include delay.h and get "hardware/adafruit/Adafruit_nRF52_Arduino/cores/nRF5/common_func.h:45:7: error: template with C linkage"

It looks like delay includes rtos.h from inside an extern "C" block, which in turn includes common_func.h which due to:
`#ifdef __cplusplus
  // namespace and templates must be outside the `extern "C"` declaration...
  namespace ADAFRUIT_DETAIL
  {
      template <typename T, size_t N>
      constexpr size_t arrcount_fails_if_not_array(T const (&)[N]) noexcept
      {   
          return N;
      }
  }
  extern "C" {
  #define arrcount(arr) ADAFRUIT_DETAIL::arrcount_fails_if_not_array(arr)
`
really doesn't want to be included from within an extern "C" block. Currently I'm working around it by including stddef.h and common_func.h before delay.h to keep the template from happening inside the extern "C"

I'm hinestly not sure if this should be called a bug, infelicity, or programmer's note, but it doesn't seem unreasonable to include delay.h in a library, and depending on common_func.h just happening to normally be included before getting re-included from inside extern "C" doesn't seem right.

hathach, 2021-07-12:
@pyro9 could you please use the bug template to report issue

pyro9, 2021-07-12:
@hathach This issue does not really fit into the bug template. I have cleaned up the format of the message.

hathach, 2021-07-13:
I am appreciate your effort to open issue, but all bug report must be filled with bug template form. If this is not a bug report, please open an topic in support forum.

closed 2021-07-13 by hathach