adafruit/Adafruit_nRF52_Arduino #676: Adafruit nRF52 0.24.0 library fails to compile on Windows 7

xchg-dot-ca, 2021-07-19:
### Operating System

Windows 7

### IDE version

1.8.15

### Board

Adafruit ItsyBitsy nRF52840 Express

### BSP version

BN: Adafruit ItsyBitsy nRF52840 Express VID: 239A PID: 8051 SN: 0635AC83ACFC813F

### Sketch

Any sketch

### What happened ?

Updated to the latest library 0.24.0
Failed to compile ANY sketch with error 
Error loading Python DLL 'C:\Users\misha\AppData\Local\Temp\_MEI67922\python39.dll'.
Windows 7 does not support Python 3.9
0.21.0
Works fine

### How to reproduce ?

Use library 0.24.0 and compile any sketch using Adafruit ItsyBitsy nRF52840 Express board.

### Debug Log

```
Compiling core...
"C:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wno-pointer-arith -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=64000000 -DARDUINO=10815 -DARDUINO_NRF52840_ITSYBITSY -DARDUINO_ARCH_NRF52 "-DARDUINO_BSP_VERSION=\"0.24.0\"" -DNRF52840_XXAA -DARDUINO_NRF52_ITSYBITSY -DUSBCON -DUSE_TINYUSB -DUSB_VID=0x239A -DUSB_PID=0x8051 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"ItsyBitsy nRF52840 Express\"" -DSOFTDEVICE_PRESENT -DARDUINO_NRF52_ADAFRUIT -DNRF52_SERIES -DDX_CC_TEE -DLFS_NAME_MAX=64 -Ofast -DCFG_DEBUG=0 -DCFG_LOGGER=1 -DCFG_SYSVIEW=0 "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS\\5.7.0/CMSIS/Core/Include/" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\CMSIS\\5.7.0/CMSIS/DSP/Include/" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/nordic" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/nordic/nrfx" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/nordic/nrfx/hal" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/nordic/nrfx/mdk" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/nordic/nrfx/soc" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/nordic/nrfx/drivers/include" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/nordic/nrfx/drivers/src" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include/nrf52" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/freertos/Source/include" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/freertos/config" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/freertos/portable/GCC/nrf52" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/freertos/portable/CMSIS/nrf52" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/sysview/SEGGER" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/sysview/Config" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0/libraries/Adafruit_TinyUSB_Arduino/src/arduino" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5" "-IC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\variants\\itsybitsy_nrf52840_express" "C:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\variants\\itsybitsy_nrf52840_express\\variant.cpp" -o "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\core\\variant.cpp.o"
Using precompiled core: C:\Users\misha\AppData\Local\Temp\arduino_cache_818571\core\core_adafruit_nrf52_itsybitsy52840_softdevice_s140v6,debug_l0_82c47f9858987b123021d640dca68f7f.a
Linking everything together...
"C:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4/bin/arm-none-eabi-gcc" "-LC:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437" -Ofast -Wl,--gc-sections -save-temps "-LC:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0\\cores\\nRF5/linker" -Tnrf52840_s140_v6.ld "-Wl,-Map,C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437/SmartServoArm-IK.ino.map" -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -u _printf_float -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align --specs=nano.specs --specs=nosys.specs -o "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437/SmartServoArm-IK.ino.elf" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\sketch\\SmartServoArm-IK.ino.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Fabrik2D\\FABRIK2D.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Wire\\Wire_nRF52.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\tusb.c.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\arduino\\Adafruit_TinyUSB_API.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\arduino\\Adafruit_USBD_CDC.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\arduino\\Adafruit_USBD_Device.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\arduino\\hid\\Adafruit_USBD_HID.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\arduino\\midi\\Adafruit_USBD_MIDI.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\arduino\\msc\\Adafruit_USBD_MSC.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\arduino\\ports\\esp32\\Adafruit_TinyUSB_esp32.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\arduino\\ports\\nrf\\Adafruit_TinyUSB_nrf.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\arduino\\ports\\rp2040\\Adafruit_TinyUSB_rp2040.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\arduino\\ports\\samd\\Adafruit_TinyUSB_samd.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\arduino\\webusb\\Adafruit_USBD_WebUSB.cpp.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\class\\audio\\audio_device.c.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\class\\bth\\bth_device.c.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\class\\cdc\\cdc_device.c.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\class\\dfu\\dfu_rt_device.c.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\class\\hid\\hid_device.c.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\class\\midi\\midi_device.c.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\class\\msc\\msc_device.c.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\class\\net\\net_device.c.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\\class\\usbtmc\\usbtmc_device.c.o" "C:\\Users\\misha\\AppData\\Local\\Temp\\arduino_build_98437\\libraries\\Adafruit_TinyUSB_Arduino\ [... cut ...]
Using library Fabrik2D at version 1.0.3 in folder: C:\Users\misha\Documents\Arduino\libraries\Fabrik2D 
Using library Wire at version 1.0 in folder: C:\Users\misha\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.24.0\libraries\Wire 
Using library Adafruit_TinyUSB_Arduino at version 1.3.0 in folder: C:\Users\misha\AppData\Local\Arduino15\packages\adafruit\hardware\nrf52\0.24.0\libraries\Adafruit_TinyUSB_Arduino 
exit status -1
Error compiling for board Adafruit ItsyBitsy nRF52840 Express.
```

hathach, 2021-07-19:
the issue with python dll is weird. How do you install python on windows. There is reported issues when installed from windows store. It is best to download and install it from https://www.python.org/downloads/windows/ .

xchg-dot-ca, 2021-07-19:
On May 14th, 2021, downloaded python-3.8.10-amd64 and installed this executable, no extra steps were performed since then

hathach, 2021-07-19:
Are you download from python.org link above ? there is probably issue with your python installation, you mentioned installing 3.8 but the error complained about python 3.9. Could you try to uninstall and reinstall with latest python.

xchg-dot-ca, 2021-07-19:
Yes this binary if from https://www.python.org/downloads/windows/
And yes I did reinstall it, w/o any success and I never had or  possible could install 3.9 or Win7 machine.
Also library version 0.21.0 works fine.

hathach, 2021-07-20:
Ah, I see, the release of recent adafruit-nrfutil is packaged with python 3.9 in mind and cause issue on windows 7 which is only 3.8. can you try to download an old version of exe from nrfutil page here probably (0.5.3post13 or post10)  https://github.com/adafruit/Adafruit_nRF52_nrfutil/releases and replace with the one come with your bsp in . Let me know which version work for you, I will try to see if we could reduce the python requires to 3.8 on windows binaries in the next release. 

```
"C:\\Users\\misha\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\nrf52\\0.24.0/tools/adafruit-nrfutil/win32/adafruit-nrfutil.exe"
```

SSIinc, 2021-07-29:
I was having the same issue and was solved by replacing .exe with the post10 as you suggested!

hathach, 2021-07-30:
thanks @SSIinc for your feedback, I will limit the python to 3.8 for it to compatible with win7 and roll out an new release to fix this.

hathach, 2021-07-30:
@SSIinc I made an draft release here that use python 3.8 to generate windows binaries https://github.com/hathach/Adafruit_nRF52_nrfutil/releases/tag/0.5.3post17rc

Could you mind trying to test it out, if worked, we will include it in the next bsp release.

SSIinc, 2021-07-30:
Works well @hathach .  Compiled and uploaded successfully with Feather Express and CLUE. Thanks for all you do!

hathach, 2021-07-30:
Thanks for confirmation, we have release post17 version where windows binary is forced to build with python 3.8. It will be bundled with the next release of this BSP. Meanwhile please manually download and replace it.

PR #681 (2021-07-31): update nrfutil binary to post17 for compatible with win7
win7 can only install up to python 3.8. this pr update windows binary which is rebuilt with python 3.8 (previously 3.9) to fix #676
closed 2021-07-30 by hathach