adafruit/Adafruit_nRF52_Arduino #679: Add the ability to set the SAADC sample time - Patches provided

ajs123, 2021-07-29:
The [Nordic nrf52 Product Specification for the ADC](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Fsaadc.html) describes the recommended relationship between the source resistance of an analog signal source and the sample Acquisition Time (see Figure 6 and the accompanying chart of TACQ values).

These changes provide the ability to set the TACQ field in the ADC channel configuration, for users who are measuring voltages from higher-impedance sources. The needed TACQ field #defines are already provided in nrf52840_bitfields.h. In testing, the standard deviation of 10-bit ADC readings at the wiper fo a 20 kohm potentiometer was reduced from 1.5 LSB to 0.7 LSB by changing TACQ from 3us to 10us.

The added function, analogSampleTime(uint8_t time) defaults to the current value of 3 us. I followed the same form as analogReadResolution and analogOversample, both of which silently set the default if not provided with a valid value.

These changes are against v0.24.  If there's a better way to provide the patches, I'd be happy to provide them in the preferred form.

```
diff wiring_analog.h wiring_analog.h.ORIGINAL
110,118d109
< /*
<  * \brief Set the ADC sample time. Default is 3us (appropriate for low source resistance).
<  * See the chart of appropriate sample time vs. source resistance in the SAADC section
<  * of the Nordic nrf52 product specification.
<  *
<  * \param time Should be set to 3, 5, 10, 15, 20 or 40.
<  */
< extern void analogSampleTime(uint8_t sTime);
< 
```

```
diff wiring_analog_nrf52.c  wiring_analog_nrf52.c.ORIGINAL
33d32
< static uint32_t saadcSampleTime = SAADC_CH_CONFIG_TACQ_3us;
144,165d142
< void analogSampleTime( uint8_t sTime)
< {
<   saadcSampleTime = SAADC_CH_CONFIG_TACQ_3us; // default is 3 us
<   switch (sTime) {
<     case 5:
<       saadcSampleTime = SAADC_CH_CONFIG_TACQ_5us;
<       break;
<     case 10:
<       saadcSampleTime = SAADC_CH_CONFIG_TACQ_10us;
<       break;
<     case 15:
<       saadcSampleTime = SAADC_CH_CONFIG_TACQ_15us;
<       break;
<     case 20:
<       saadcSampleTime = SAADC_CH_CONFIG_TACQ_20us;
<       break;
<     case 40:
<       saadcSampleTime = SAADC_CH_CONFIG_TACQ_40us;
<       break;
<   }
< }
< 
197c174
<                             | ((saadcSampleTime               << SAADC_CH_CONFIG_TACQ_Pos)   & SAADC_CH_CONFIG_TACQ_Msk)
---
>                             | ((SAADC_CH_CONFIG_TACQ_3us      << SAADC_CH_CONFIG_TACQ_Pos)   & SAADC_CH_CONFIG_TACQ_Msk)
```

hathach, 2021-07-30:
could you submit this as PR, it is much easier to discuss/review code change with PR than issue.

ajs123, 2021-07-30:
Done. Thanks!

closed 2021-07-30 by ajs123