adafruit/Adafruit_nRF52_Arduino #682: Add a function to read the on-board temperature sensor

ajs123, 2021-08-02:
Unless I've missed something, there's the ability to read the on-chip temperature sensor in Circuit Python but not in Arduino.  It appears to be a simple matter.

Nordic recommends calibration of the SAADC offset occasionally or if the temperature changes more than 10C. I've written the offset calibration and will add it to the analog input code and submit as a PR.  

Should I add the reading of the on-chip temp there as well, or put it elsewhere?  I'll put it wherever is preferred and submit as a PR.

Thanks!

hathach, 2021-08-03:
yeah, thank you, that would be great, you could add both of the code within the PR since you need the cpu temperature anyway. For the API name, I am not entire sure how other core use it, maybe `readCPUTemperature()` would be great. For location it is probably be placed at wiring.h/c https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/wiring.h . However, if you figure out an better place and better name, go for it, we could discuss more in the PR.

hathach, 2021-08-10:
implemented by #683

closed 2021-08-10 by hathach