adafruit/Adafruit_nRF52_Arduino #693: Not able to enable notification on a custom characteristic (solved)

BigCorvus, 2021-09-17:
### Operating System

Windows 10

### IDE version

Arduino 1.8.16

### Board

Feather nRF52840 express

### BSP version

1.0.0

### Sketch

Adafruit_nRF52_Arduino/libraries/Bluefruit52Lib/examples/Central/central_custom_hrm/central_custom_hrm.ino

modified to connect with custom service and custom characteristic of a moodmetric smart ring

    const uint8_t MOODMETRIC_UUID_SERVICE[16] =
    {
      0x8E, 0x3F, 0x28, 0xB7, 0xAA, 0xA7, 0x23, 0xA9,
      0x88, 0x49, 0xCD, 0xE4, 0x70, 0x9B, 0x49, 0xDD
    };

    const uint8_t MOODMETRIC_UUID_CHR_STREAMING[16] =
    {
      0x66, 0x9A, 0x0C, 0x20, 0x00, 0x08, 0x06, 0xBD,
      0xE4, 0x11, 0xD2, 0x9B, 0x20, 0x64, 0x95, 0xA0
    };

### What happened ?

Notifications on the Ring could not be enabled by enableNotify() until I changed the .write_op = BLE_GATT_OP_WRITE_CMD to .write_op = BLE_GATT_OP_WRITE_REQ
in the BLEClientCharacteristic::writeCCCD() of BLEClientCharacteristic.cpp.

Found the inspiration to the solution here: https://devzone.nordicsemi.com/f/nordic-q-a/40602/unable-to-enable-notifications-using-sd_ble_gattc_write-sdk-15-2

### How to reproduce ?

Just in case notifications on a peripheral are not enabled: this change in the ble_gattc_write_params_t param of the function writeCCCD() should be attempted.

### Debug Log

No errors. writeCCCD() worked, but the notification callback was never called until I changed this one value.

ble_gattc_write_params_t param =
  {
      .write_op = BLE_GATT_OP_WRITE_REQ, //formerly BLE_GATT_OP_WRITE_CMD
      .flags    = BLE_GATT_EXEC_WRITE_FLAG_PREPARED_WRITE,
      .handle   = _cccd_handle,
      .offset   = 0,
      .len      = 2,
      .p_value  = (uint8_t*) &value
  };

hathach, 2021-09-18:
thank for filing this issue, could you be more specific on the issue and how to reproduce it. Thing can be different on different app or mobile OS or even phone.
1. please attached your sketch with the modification for reproducing. Also please add comments explain where the the issue e.g in the callback. maybe add printf() for both expected behavior.  
2. In `How to reproduce` section please specify exact steps to reproduce the issue: which app that you use, your mobile OS version etc, and which steps within the app that trigger the issue. And what is expected behavior etc..
3. You still need to provide the debug log (level 2) as the attached txt around the event where the incident occurs for analyzing the software behavior.

hathach, 2021-11-11:
closed since user marks it as solved.

closed 2021-11-11 by hathach

kushalsoniks, 2024-04-26:
This saved me!! I can confirm I had the same issue with the Chipolo One Beacon.