adafruit/Adafruit_nRF52_Arduino #695: Add ability to read Vdd: Looking for preference in how to code the pull request

ajs123, 2021-10-02:
For my application, I'd like to be able to read Vdd.  The nrf52 allows this by selecting SAADC channel 9.  

I see three options for where to put this:

1. Add readVdd() to wiring_analog_nrf52.c. 
2. Add a pin for Vdd so that it can be selected in analogRead().
3. Make analogReadInternal() public in wiring_analog_nrf52.h so that channel 9 can be chosen.

I think the best way is #1 but I wanted to check before adding the code. I'll follow with a PR.

Thanks!

lyusupov, 2021-10-03:
> 1. Add readVdd() to wiring_analog_nrf52.c.

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/7770246e2590f4e0b81676c4d1a199ab2055f9f6/cores/nRF5/wiring_analog_nRF52.c#L282

ajs123, 2021-10-03:
Never mind - I now see that analogReadVDD is already there!

closed 2021-10-03 by ajs123