adafruit/Adafruit_nRF52_Arduino #696: Support for nRF52833 (through PCA10100 - nRF52833 DK)

jpconstantineau, 2021-10-08:
Would support for the nRF52833 DK be merged in if one would develop it?

I have a fork of the project that adds boards and would rather have support from the core for compiling towards the 833 and bringing support for the Dev Kit upstream would make it much easier to merge and keep the fork in sync.

dhalbert, 2021-10-08:
Yes, if you submit a PR, we would be happy to add this board

dhalbert, 2021-10-08:
Sorry, I thought this issue was in the circuitpython repo. @ladyada, would we add PCA10100? From the README:

> Following boards are also included but are not officially supported:
> 
> - Nordic nRF52840DK PCA10056
> - Particle Xenon

jpconstantineau, 2022-01-26:
Any decision on whether support for the 833 would be merged in if a PR would be submitted?  #698 is WIP but opens up the way for people to work with the 833 using Arduino and potentially SoftDevice 7 support/updates.

LokiMetaSmith, 2022-11-21:
Just a note, I got a version of the bootloader and this working. It's not in any particular order, but here's my work in progress. 
https://github.com/LokiMetaSmith/Adafruit_nRF52_Bootloader
https://github.com/LokiMetaSmith/Adafruit_nRF52_Arduino/tree/nrf5233_compliesbutneedsbootloader

I got really lost at one point because it is necessary to update the bootloader to get nrf52833 to work correctly, and there was a bit of chasing my own shadow, trying to figure out what all needed to be changed to get it to work with the new part. 

Another angle I was working down was replacing the bluefruit bluetooth stack with NimBLE, which removed the need for Nordic SoftDevices all together. I got most of the way there, but got hung up because I had utilized many of the helper functions in the bluefruit stack, and NimBLE had just different enough of an API that it wasn't a drop in replacement. 
https://github.com/LokiMetaSmith/n-able-Arduino

Mboudali, 2023-07-31:
up.

oldmud0, 2023-08-25:
Hi @LokiMetaSmith - thanks a lot for the '833 port! I am trying to port a program from an '840 (XIAO BLE) to a custom '833 board and it turned out to be much more daunting than expected as there isn't complete support for it via Adafruit. Currently I combined various changes in your fork and another fork that I found to add support, but for whatever reason, the firmware doesn't succeed getting out of `vPortStartFirstTask` after invoking the service handler and ends up spinning in SoftDevice territory (s113 7.3.0).

```
Reading symbols from ~/projects/wf_egg/firmware_xiao/.pio/build/myboard/firmware.elf...done.
(gdb) load
Loading section .text, size 0x37690 lma 0x27000
Loading section .ARM.exidx, size 0x8 lma 0x5e690
Loading section .data, size 0x2e0 lma 0x5e698
Start address 0x38f7c, load size 227704
Transfer rate: 4834 KB/sec, 13394 bytes/write.
(gdb) bt
#0  Reset_Handler () at /home/me/.platformio/packages/framework-arduinoadafruitnrf52/cores/nRF5/linker/gcc_startup_nrf52833.S:232
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
0x0001ada0 in ?? ()
(gdb) bt
#0  0x0001ada0 in ?? ()
#1  <signal handler called>
#2  0x0001aec2 in ?? ()
#3  <signal handler called>
#4  0x00038ef2 in vPortStartFirstTask () at /home/me/.platformio/packages/framework-arduinoadafruitnrf52/cores/nRF5/freertos/portable/GCC/nrf52/port.c:53
#5  0x00038c26 in xPortStartScheduler ()
    at /home/me/.platformio/packages/framework-arduinoadafruitnrf52/cores/nRF5/freertos/portable/CMSIS/nrf52/port_cmsis.c:240
#6  0x00000000 in ?? ()
```

Did you encounter this issue, and based on your knowledge is it required for me to flash the bootloader as well? I thought I could just flash the firmware directly over J-Link, run it, and deal with the bootloader later.

Thanks in advance.