adafruit/Adafruit_nRF52_Arduino #700: Doen`t work SPIM3 on nRF52840

Fantonrko, 2021-10-17:
### Operating System

Windows 10

### IDE version

1.8.16

### Board

Feather nRF52840 Express

### BSP version

1.0.1

### Sketch

#define SPI_INTERFACES_COUNT 2
#define SPI_32MHZ_INTERFACE 1
#define PIN_SPI_MISO (30)
#define PIN_SPI_MOSI (39)
#define PIN_SPI_SCK (37)

#ifndef SPI_32MHZ_INTERFACE 1
#define SPI_32MHZ_INTERFACE 1
#endif
#if SPI_32MHZ_INTERFACE == 0
  #define _SPI_DEV    NRF_SPIM3 // 32 Mhz
  #define _SPI1_DEV   NRF_SPIM2
#elif SPI_32MHZ_INTERFACE == 1
  #define _SPI_DEV    NRF_SPIM2
  #define _SPI1_DEV   NRF_SPIM3 // 32 Mhz
#else
  #error "not supported yet"
#endif
#if SPI_INTERFACES_COUNT >= 1
SPIClass SPI(_SPI_DEV,  PIN_SPI_MISO,  PIN_SPI_SCK,  PIN_SPI_MOSI);
#endif
#if SPI_INTERFACES_COUNT >= 2
SPIClass SPI1(_SPI1_DEV, PIN_SPI1_MISO, PIN_SPI1_SCK, PIN_SPI1_MOSI);
#endif

### What happened ?

Have enable SPIM3 like sketch, But SPI speed still 8MHz on my oscilloscope. What`s i can do to use max 32MHz ?

I already reported this issue on the Adafruit forum, but no one there was able to help in the correct SPIM3 setup and use. I have tried changing various parameters in the libraries that refer to changing the speed, but never got the desired result

### How to reproduce ?

-

dhalbert, 2021-10-17:
Could you show the whole sketch, or attach it as a .txt file? This does not include `PIN_SPI1_MISO`, or any use of the SPI.

dhalbert, 2021-10-17:
Please show your whole sketch. That is only part of your Arduino sketch.

Fantonrko, 2021-10-17:
> Please show your whole sketch. That is only part of your Arduino sketch.

U mean configuration files ( variant.h, spi.h, spi.cpp and etc.) or only arduino sketch with tft example ?

dhalbert, 2021-10-17:
The sketch only, but the whole sketch. We have `variant.h`, `spi.cpp` etc.,assuming they are what is in the released board support package. 1.1.0 is actually available, and you are using 1.0.0, but I don't think there are changes that affect this.

dhalbert, 2021-10-17:
You can upload it as a file (zip it or change the extension), instead of pasting it in.

Fantonrko, 2021-10-17:
> The sketch only, but the whole sketch. We have `variant.h`, `spi.cpp` etc.,assuming they are what is in the released board support package. 1.1.0 is actually available, and you are using 1.0.0, but I don't think there are changes that affect this.

oh, sorry, i use 1.0.1 version.  I use 2 library for graphic, Arduino_GFX and TFT_eSPI (prefer second). If u need user_Setup from TFT_eSPI i can upload it. So i use Hardware SPI ping, which I wrote in variant.h.
[TFT_Test.zip](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/7360981/TFT_Test.zip)

Fantonrko, 2021-10-18:
> > @Fantonrko for a bug report, you need upload all needed codes so that we could open it and reproduce the issue. Any out of code configure like variant modification or hardware setup need to be stated clearly. The codes should be as **minimal** as possible to reduce the scope of analysis. I just check your uploaded example, it is too complicated for a bug report. Please remove all the code that is not needed to reproduce the issue and try again.

па

Fantonrko, 2021-10-19:
`паапарп`

dhalbert, 2021-10-19:
I'm confused. You had to make some modifications to get it to do 32MHz. Try a simpler example that doesn't use https://github.com/moononournation/Arduino_GFX. Just create a regular Arduino SPI object, send something on it, and look at the clock rate. Don't use any libraries.