adafruit/Adafruit_nRF52_Arduino #701: LittleFS boot_count example

milador, 2021-11-22:
### Operating System

Windows 10

### IDE version

Arduino 1.8.16

### Board

Feather nRF52840 Sense

### BSP version

1.1.0

### Sketch

The boot_count example from [littlefs readme](https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Adafruit_LittleFS/src/littlefs) file.

### What happened ?

No information on how to implement block device operations and the main configuration. Changed user_provided_block_device_read to lfs_bd_read from `lfs.h` and I am getting errors .

`error: 'lfs_bd_read' was not declared in this scope; did you mean 'lfs_dir_read'?`

I would like to read and write to the flash.

### How to reproduce ?

- Copy and paste the sample code from [littlefs readme](https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Adafruit_LittleFS/src/littlefs)
-  Reference `lfs.h`

hathach, 2021-11-22:
the LFS is already used to manage bonding, there is an higher library that abstract the LFS API to familiar SD file APII
- https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/InternalFileSytem/src/InternalFileSystem.cpp 

you could follow its example there
-  https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/InternalFileSytem/examples

closed 2021-11-22 by hathach

milador, 2021-11-22:
Can it be used to achieve something similar to EEPROM or FlashStorage? I basically want to read and write variables  instead of the intire 128 bytes block of data every time. Thanks

hathach, 2021-11-22:
yes, it does, check out its example

milador, 2021-11-24:
@hathach Thanks for pointing me toward right direction.I followed the example you provided. But can't figure out how to read and write without creating a file. I must create a file for this approach , right? I basically need to write few integers as device configuration.

hathach, 2021-11-24:
LFS is a file system, everything is a file. you should use raw flash API if it is not what you are looking for.

milador, 2021-11-24:
@hathach I see, any example for raw flash API read/write ? Something like `flash_nrf5x_write(addr, buffer, size)` ?

hathach, 2021-11-24:
For questions, please post it on the adafruit support forum. There is more people there to help you out, github is mostly for troubleshooting