adafruit/Adafruit_nRF52_Arduino #713: Trying to add support for Makerdiary nRF52840 MDK Dongle

Brezensalzer, 2022-01-29:
Hi,
I'm trying to add a new board variant (Makerdiary nRF52840 MDK Dongle) to Adafruit_nRF_Arduino.
What I have done so far:
* added new variant.h/variant.cpp files to variants/mdk_nrf52840_dongle
* copied the bootloader files from [here](https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases) to bootloader/mdk_nrf52840_dongle (Version 0.6.2). This Bootloader is used for CircuitPython
* added the board definition to boards.txt
* flashed the bootloader via Arduino IDE (J-Link or DFU, both works)
* a USB disk "MDK840DONGL" pops up and the serial USB port enumerates
* I can compile sketches in the Arduino IDE for this board error free

But the Upload always fails with:
```
Library Adafruit_nRFCrypto has been declared precompiled:
Using precompiled library in /home/stl/.arduino15/packages/adafruit/hardware/nrf52/1.3.0/libraries/Adafruit_nRFCrypto/src/cortex-m4/fpv4-sp-d16-hard
Sketch uses 45232 bytes (5%) of program storage space. Maximum is 815104 bytes.
Global variables use 13672 bytes (5%) of dynamic memory, leaving 223896 bytes for local variables. Maximum is 237568 bytes.
Upgrading target on /dev/ttyACM0 with DFU package /tmp/arduino_build_719582/serialOut.ino.zip. Flow control is disabled, Single bank, Touch disabled

Failed to upgrade target. Error is: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 500, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/stl/.local/lib/python3.8/site-packages/nordicsemi/__main__.py", line 296, in serial
    dfu.dfu_send_images()
  File "/home/stl/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 235, in dfu_send_images
    self._dfu_send_image(HexType.APPLICATION, self.manifest.application)
  File "/home/stl/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 203, in _dfu_send_image
    self.dfu_transport.send_init_packet(init_packet)
  File "/home/stl/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 155, in send_init_packet
    self.send_packet(packet)
  File "/home/stl/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 243, in send_packet
    ack = self.get_ack_nr()
  File "/home/stl/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 264, in get_ack_nr
    temp = [x for x in self.serial_port.read(6)]
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

Possible causes:
- Selected Bootloader version does not match the one on Bluefruit device.
    Please upgrade the Bootloader or select correct version in Tools->Bootloader.
- Baud rate must be 115200, Flow control must be off.
- Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.

```

I'm a newbie, so I have no idea what could be wrong. The bootloader version should be correct. 
Here is the content of INFO_UF2.TXT:
```
_UF2 Bootloader 0.6.2 lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-293-gaf8e5a90) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: MDK nRF52840 USB Dongle
Board-ID: nRF52840-Dongle-v1
SoftDevice: S140 version 6.1.1
Date: Sep 10 2021_

```
Has someone a hint for me?
Thanks in advance

jpconstantineau, 2022-01-30:
1 - have you tried "double-reset" to make sure that it's in bootloader mode and select the serial port that the bootloader provides?
2 - Since it's a non-Adafruit board, I doubt that you will be able to merge it in the source repo.  I suggest you add it to the Community nRF52 repo (https://github.com/jpconstantineau/Community_nRF52_Arduino)

Brezensalzer, 2022-01-30:
@jpconstantineau 
Thank You for Your reply
1 - I did that, but it made no difference
2 - The board is officially supported by CircuitPython and there is a working bootloader from Adafruit available. It should be really no problem to add the board with a pull request. There are already other non-Adafruit boards in this repo too. But I will have a look at the Community nRF52 repo.

ladyada, 2022-01-30:
hiya we wouldn't add makerdiary to this repo because we don't own or stock that item - its not possible for us to maintain it. the community repo is definitely the place for it. you can also ask makerdiary to maintain a package for the hardware they sell, that would be ideal!
thank you :)

Brezensalzer, 2022-01-31:
Hi
ok, than I will try out the community repo.
Thank You!

closed 2022-02-01 by hathach