adafruit/Adafruit_nRF52_Arduino #714: https://www.adafruit.com/package_adafruit_index.json page not found

yoursunny, 2022-01-29:
### Operating System

Linux

### IDE version

arduino-cli  Version: 0.20.2 Commit: 13783819 Date: 2021-12-09T13:32:05Z

### Board

nRF52832

### BSP version

N/A

### Sketch

N/A

### What happened ?

I'm trying to install Arduino Core for nRF52 boards via command line, but the installation fails because the JSON file does not exist.

### How to reproduce ?

```bash
arduino-cli core update-index --additional-urls https://www.adafruit.com/package_adafruit_index.json
```

### Debug Log

https://github.com/yoursunny/NDNph/runs/4992915148?check_suite_focus=true

dhalbert, 2022-01-29:
Use this URL instead: https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

Where did see the incorrect https://www.adafruit.com/package_adafruit_index.json URL? We had reports of this, and I fixed several mentions of this incorrect URL yesterday in the Learn Guides, including in screenshots.

yoursunny, 2022-01-29:
> Where did see the incorrect https://www.adafruit.com/package_adafruit_index.json URL?

README.md of this repository.

If changing URI, please setup 301 redirect from the old URI.

ladyada, 2022-01-30:
fixed the readme, please use the new URL only https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

closed 2022-01-30 by ladyada