adafruit/Adafruit_nRF52_Arduino #715: No MIDI.h library

mbmosher, 2022-02-04:
The blemidi example file needs an update.  

The link in the file's comments (https://learn.adafruit.com/wireless-untztrument-using-ble-midi/overview) results in a 404 error.  I believe it should be updated to: https://learn.adafruit.com/wireless-untztrument-using-ble-midi/overview-uniontownlabs

Following that guide installs the libraries:
    [Adafruit Trellis](https://github.com/adafruit/Adafruit_Trellis_Library)
    [Adafruit BluefruitLE nRF51](https://github.com/adafruit/Adafruit_BluefruitLE_nRF51)
    [Adafruit UNTZtrument](https://github.com/adafruit/Adafruit_UNTZtrument) v2.0.0 or higher

However, compiling results in the error "MIDI.h: no such file or directory'

I also tried installing the MIDIUSB library, no change.  Where does the MIDI.h library come from?

hathach, 2022-02-04:
> The blemidi example file needs an update.
>
> The link in the file's comments (https://learn.adafruit.com/wireless-untztrument-using-ble-midi/overview) results in a 404 error. I believe it should be [...]

Thanks for the issue, this learn tutorial is mostly for host setup, the board in tutorial is nrf51 series while CircuitPlay Bluefruit is a nrf52 board. Please check out this learn guide for installing midi lib dor nrf52 series https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/blemidi

I will update the link and description in the example to prevent confusion

ladyada, 2022-02-04:
thank you @hathach 🙏

closed 2022-02-18 by hathach