adafruit/Adafruit_nRF52_Arduino #716: blufruit.h will not compile - error with hid_gamepad_report_t

vader7071, 2022-02-06:
### Operating System

Windows 10

### IDE version

Both 1.8.15 and 2.0.0-rc2

### Board

nRF52832

### BSP version

github latest as of 02/04/2022

### Sketch

/*
 * 2022-02-04
 * 
 * Adafruit Feather nRF52
 * nRF52832
 * 
 * Adafruit Products:
 *   3090 (0.8" 8x8 Matrix)
 *   3406 (Adafruit Feather nRF52 Bluefruit LE - nRF52832)
 *   
 *   
 * Does 4 things
 *   1)  Scrolls the message on (1) Adafruit 3090 16x8 mini matrix
 *   2)  Stacks 6 leds on back of helmet
 *   3)  Receives data via Bluetooth
 *   4)  Positions the Range Finder Servo
 */

//============  Includes / Defines  ============
#include <Wire.h>
//#include "D:\Documents\Arduino\libraries\Servo\src\Servo.h" - This version does not cause "multiple found" report.
//#include "Servo.h"
#include <Adafruit_LEDBackpack.h>
#include <Adafruit_GFX.h>
#include <bluefruit.h>
#include <SPI.h>

#define NUM_MATRICES 1

//============  Constants  ============
// --------- Digital Pins ---------
int led_Pin_0 = 2;            // Dig I/O  PWM
int led_Pin_1 = 3;            // Dig I/O  PWM
int led_Pin_2 = 4;            // Dig I/O  PWM
int led_Pin_3 = 5;            // Dig I/O  PWM
// int rf_servo = 16;            // Dig I/O  PWM
int led_Pin_4 = 28;           // Dig I/O  PWM
int led_Pin_5 = 29;           // Dig I/O  PWM

//============  Module Configuration  ============
// --------- BLE Setup ---------
// BLE Service
BLEDfu  bledfu;  // Over The Air Device Firmware Update service
BLEDis  bledis;  // Device Information Service
BLEUart bleuart; // uart over ble
BLEBas  blebas;  // BAttery Service

// --------- Servo ---------
// Servo rfServo;

// --------- 16x8 Mini Matrix ---------
Adafruit_8x16minimatrix matrix = Adafruit_8x16minimatrix();

//--------  Sets scrolling messages  --------
char* msg[] = {"                ","Suck it Wolverine!","Free Tacos      ","Don't be a Dick!"};   // Scrolling text

const int msgArraySize = 4;                                      // Sets the number of spots in the arrays
const int ledArraySize = 6;                                      // Sets the number of spots in the arrays

// ---- Sets Back LED Timer Multiplier ----
const int offset = 15;                                           // delay time for each LED

// ---- Used for Scroll Speed ----
const int msgInterval = 45;                                      // number of millisecs between blinks

//============  VARIABLES (will change)  ============
int msgChoice = 0;                                               // holds the selection for which message to show
int mark = 0;                                                    // for LED stacking
int spot = 0;                                                    // for LED stacking

// ---- Loop Timing Counters ----
int scroll_loop = 0;
int ledStack_loop = 0;

//============  Arrays  ============
int led_Pin[6] = {led_Pin_0, led_Pin_1, led_Pin_2, led_Pin_3, led_Pin_4, led_Pin_5}; 
uint8_t ble_data[4] = {0,0,0,0};

//============  Setup function  ============
void setup()
{
  Serial.begin(115200);

  Bluefruit.autoConnLed(true);

  Bluefruit.configPrphBandwidth(BANDWIDTH_MAX);

  Bluefruit.begin();
  Bluefruit.setTxPower(-4);    // Set max power. Accepted values are: -40, -30, -20, -16, -12, -8, -4, 0, 4
  Bluefruit.setName("Helmet");
 
//  Bluefruit.Periph.setConnectCallback(connect_callback);
//  Bluefruit.Periph.setDisconnectCallback(disconnect_callback);

  // To be consistent OTA DFU should be added first if it exists
  bledfu.begin();

  // Configure and Start Device Information Service
  bledis.setManufacturer("Adafruit Industries");
  bledis.setModel("nRF52832");
  bledis.begin();

  // Configure and Start BLE Uart Service
  bleuart.begin();

  // Start BLE Battery Service
  blebas.begin();
  blebas.write(100);

  // Set up and start advertising
  startAdv();

  // Setup the servo
  // rfServo.attach(rf_servo);

  for(uint8_t m=0; m<NUM_MATRICES; m++) {                      // Set the address and mode for the 8x8 matrices
    matrix.begin(0x70 + m);
    matrix.clear();
    matrix.setTextSize(1);
    matrix.setTextWrap(false);                              // we dont want text to wrap so it scrolls nicely
    matrix.setRotation(1);                                  // 0 = I2C Bottom Left; 1 = Conn @ Left; 2 = I2C Top Right; 3 = Conn @ Right
  }

  for (uint8_t n=0; n<ledArraySize; n++) {
    pinMode(led_Pin[n], OUTPUT);
  }
}

//============  Advertising function  ============
void startAdv(void) {
  // Advertising packet
  Bluefruit.Advertising.addFlags(BLE_GAP_ADV_FLAGS_LE_ONLY_GENERAL_DISC_MODE);
  Bluefruit.Advertising.addTxPower();

  // Include bleuart 128-bit uuid
  Bluefruit.Advertising.addService(bleuart);

  // Secondary Scan Response packet (optional) since there is no room for 'Name' in Advertising packet
  Bluefruit.ScanResponse.addName();

  Bluefruit.Advertising.restartOnDisconnect(true);
  Bluefruit.Advertising.setInterval(32, 244);    // in unit of 0.625 ms
  Bluefruit.Advertising.setFastTimeout(30);      // number of seconds in fast mode
  Bluefruit.Advertising.start(0);                // 0 = Don't stop advertising after n seconds  
}

//============  Loop function  ============
void loop() {
  // ---- Checks for Data from BLE ----
  while ( bleuart.available() ) {
    for (int x = 0; x < 4; x++) {
      ble_data[x] = (uint8_t) bleuart.read();
    }
    msgChoice = ble_data[0];
  }

  // ---- Calls Mesage Loop ----
    scroll(msg[msgChoice]); 
}

//============  Callback function  ============
// callback invoked when central connects
void connect_callback(uint16_t conn_handle) {
  // Get the reference to current connection
  BLEConnection* connection = Bluefruit.Connection(conn_handle);

  char central_name[32] = { 0 };
  connection->getPeerName(central_name, sizeof(central_name));

  Serial.print("Connected to ");
  Serial.println(central_name);
}

//============  Disconnect Callback function  ============
void disconnect_callback(uint16_t conn_handle, uint8_t reason) {
  (void) conn_handle;
  (void) reason;

  Serial.println();
  Serial.println("Disconnected");
}

//============  Matrix Text Scroll  ============
void scroll(char* text) {
  int scrollPositions = (strlen(text) * 8) - 32;
  for (int x=(16 * NUM_MATRICES); x>=-scrollPositions; x--) {
    for(uint8_t m=0; m<NUM_MATRICES; m++) {
      matrix.clear();
      matrix.setCursor((x - (m * 8)),0);
      matrix.print(text);
      matrix.writeDisplay();
    }

  // ---- Calls LED Stacking ----
  ledStack();

  // ---- Sets Servo ----
//  if (msgChoice != 0); {
//    rfServo.write(90);
//  }
//  if (msgChoice == 0) {
//    rfServo.write(0);
//  }
  
  delay(msgInterval);
  }
}

//============  LED Stacking  ============
void ledStack() {
  mark++;
  if (mark >= offset) {
    int val = digitalRead(led_Pin[spot]);
    digitalWrite(led_Pin[spot], !val);
    if (spot >= ledArraySize) {
      spot = 0;
    } else {
      spot += 1;
    }
    mark = 0;
  }
}

### What happened ?

When I try to compile the above sketch, or any default example sketch included in the library, the compiler errors and states the below error log (BLE HID Gamepad not declared). If I comment out the #include <bluefruit.h> line and shut down the BLE portion of the sketch, the sketch compiles fine.

### How to reproduce ?

Just try to compile.

### Debug Log

In file included from d:\Documents\Arduino\libraries\Bluefruit52Lib\src/bluefruit.h:68:0,
                 from C:\Users\admn\AppData\Local\Temp\.arduinoIDE-unsaved202215-20620-w1omn1.b534o\bleuart\bleuart.ino:14:
d:\Documents\Arduino\libraries\Bluefruit52Lib\src/services/BLEHidGamepad.h:54:17: error: 'hid_gamepad_report_t' has not been declared
     bool report(hid_gamepad_report_t const* report);
                 ^~~~~~~~~~~~~~~~~~~~
d:\Documents\Arduino\libraries\Bluefruit52Lib\src/services/BLEHidGamepad.h:60:36: error: 'hid_gamepad_report_t' has not been declared
     bool report(uint16_t conn_hdl, hid_gamepad_report_t const* report);
                                    ^~~~~~~~~~~~~~~~~~~~
Compilation error: exit status 1}

hathach, 2022-02-07:
for readability an reproducible:
1. please attached the ino sketch instead of pasting the code here. 
2. Debug log should be in code/block
3. Also make sure you tried with one of the stock example e.g blehid to see if that compiles as well.

vader7071, 2022-02-07:
1) I will go back and edit and paste the actual file, I apologize.
2) I will try and fix the debug log
3) Please see under "What happened" where it states:

     When I try to compile the above sketch, _**or any default example sketch included in the library**_...

I have tried compiling multiple example sketches in the peripheral, client, and dual roles sub categories, and all result in the same compile error concerning "hid_gamepad_result_t".

hathach, 2022-02-07:
you have nothing to apologize for, it is a typical and quite intuitive to just paste the code. If git doesn't allow to upload .ino file, just rename it to .txt to upload

hathach, 2022-02-07:
did you install the bsp using git clone. Make sure you also clone its submodule as well (tinyusb)

```
git submodule update --init
```

vader7071, 2022-02-07:
Before, I have just been downloading the zip file and extracting all of the files, then copying the libraries folder into my "arduino" folder to update.

I just downloaded Git and ran the git clone to update the libraries.

Before Git, "bluefruit.h" was located in "D:\Documents\Arduino\Libraries\Bluefruit52Lib\src"

After Git, "bluefruit.h" is located in "D:\Documents\Arduino\libraries\Adafruit_nRF52_Arduino\libraries\Bluefruit52Lib\src" But to ensure the IDE got the correct libraries, I copied the Git version over into the libraries folder, so when the IDE went searching, it would find the most up to date in either the libraries root folder or in the Adafruit_nRF52_Arduino\libraries folder.

The submodule command did not work via Git [`fatal: not a git repository (or any of the parent directories): .git`] but I did go and clone the TinyUSB repository as well.

Still, after all of that, I get the same result.  

`In file included from D:\Documents\Arduino\libraries\Adafruit_nRF52_Arduino\libraries\Bluefruit52Lib\src\bluefruit.h:68:0,
                 from D:\Documents\Arduino\2022-02-04_FettPool_BLE_Helmet\2022-02-04_FettPool_BLE_Helmet.ino:25:
D:\Documents\Arduino\libraries\Adafruit_nRF52_Arduino\libraries\Bluefruit52Lib\src\services/BLEHidGamepad.h:54:17: error: 'hid_gamepad_report_t' has not been declared
     bool report(hid_gamepad_report_t const* report);
                 ^~~~~~~~~~~~~~~~~~~~
D:\Documents\Arduino\libraries\Adafruit_nRF52_Arduino\libraries\Bluefruit52Lib\src\services/BLEHidGamepad.h:60:36: error: 'hid_gamepad_report_t' has not been declared
     bool report(uint16_t conn_hdl, hid_gamepad_report_t const* report);
                                    ^~~~~~~~~~~~~~~~~~~~
Compilation error: exit status 1}`

closed 2022-02-07 by hathach