adafruit/Adafruit_nRF52_Arduino #717: compile fail on arduino ide 2.0

HeFeng1947, 2022-02-14:
### Operating System

Windows 10

### IDE version

2.0 rc3

### Board

arduino nano ble 33

### BSP version

1.3.0

### Sketch

blinky

### What happened ?

compile fail

### How to reproduce ?

At present, I have uninstalled the 2.0 ide and switched to ide version 1.8.19, which can compile normally.

ladyada, 2022-02-14:
arduino nano ble 33 isnt supported by our board package - please use arduino's board package
https://docs.arduino.cc/hardware/nano-33-ble

HeFeng1947, 2022-02-15:
> arduino nano ble 33 isnt supported by our board package - please use arduino's board package https://docs.arduino.cc/hardware/nano-33-ble

Maybe my description is not accurate? What I mean by this is that example gets an error when compiled directly, whereas in ide version 1.8.19 it gets compiled and uploaded to the board. About the board support, I replaced Adafruit_nRF52_Bootloader. 
At present, the 2.0 IDE has been deleted from my work computer, so I cannot provide more information for the time being. I will update it later after I install the environment on my presonal computer

HeFeng1947, 2022-02-15:
I reinstalled the environment of 2.0 IDE and found that the problem could not be reproduced. I guess it's because the library wasn't completely downloaded yesterday. In short, thank you for your work！

closed 2022-02-15 by HeFeng1947